egzumer/uv-k5-firmware-custom #130: Is it possible to put other options for the battery related to amperage?

pietss, 2023-11-21:
Is it possible to put other options for the battery related to amperage? 
Now the two options are mAh 1600 and 2200, can you put 3200 for example? It could be useful for those who print the body for inserting 18650 batteries https://www.thingiverse.com/thing:6162203
Thank you

egzumer, 2023-11-21:
The mAh value doesn't really matter. It could be 100000mAh. What matters is the V/percantage discharge curve. I added 2 types because those are the ones offered by QS. I traced the discharge curve for those types.
As for DIY batteries the only option is to choose the one that fits the best or to modify the curve points in the code on your own.

pietss, 2023-11-21:
ok, I understand, thank you very much.

closed 2023-11-21 by egzumer