egzumer/uv-k5-firmware-custom #136: add text messaging

mikhailsimon, 2023-11-23:
Hi, I've seen other mods that have text messaging over AFSK..do you think its possible to implement or the flash storage isn't enough

ErikS-web, 2023-11-23:
DTMF is supported in radio.
In my opinion, other digital things should be accomplished externally on a PC. And to make this possible, this issue has been submitted [See #90](https://github.com/egzumer/uv-k5-firmware-custom/issues/90)

harifhaqim, 2023-11-24:
DTMF is for calling but text messaging on QS itself is pretty handy since it's portable....DTMF does support numbers and A-D only... it's doesn't support whole alphabet and spaces

JollyH0789, 2023-11-24:
The messages are useless, better to put other more interesting functions

ErikS-web, 2023-11-24:
Out of curiosity, a few questions.
Which characters must be supported? All ISO-8859 and ISO-2022 and more icons such as smilies ;-)
Please note that everything must fit into 60KB!
The next question is the method of input.
Seems very clumsy on such a small keyboard?!

closed 2023-12-03 by ErikS-web