egzumer/uv-k5-firmware-custom #139: In Fm Radio Broadcast add more presets and if is posible RDS DATA

MachoEnLaNoche, 2023-11-24:
Hello, in FM radio when scan for new stationa it only memorize 20 radio stations, please, ad at least 100 memories or UNLIMITED.
¿IT IS POSIBLE FOR YOU TO AD "RDS" RADIO DATA SYSTEM AND MAKE TO SHOW Station Name, PTy, radio text, ct clock date and time from the radio station signal? Like in cars radios.
PS, RT, PTY, CT, AF, TP,TA...

MachoEnLaNoche, 2023-11-24:
I will be happy if you modify to accept 100 pressets
Thanks

egzumer, 2023-11-25:
This is a ham radio not a broadcast receiver. The FM radio is just an extra function. The FM radio chip has no RDS functionality and alike. 20 stations is plenty enough. I'm not willing to work on expanding that, sorry.

commit b27ef43b (2023-10-09): BUG FIX #139: always 4 battery bars
closed 2023-11-25 by egzumer