egzumer/uv-k5-firmware-custom #140: Add a option to put Personalized TEXT in Welcome screen

MachoEnLaNoche, 2023-11-24:
Hello. Will be very useful to indentify the walkie talkie in companys or groups o personal use owner.
To put an option to enable add your Own text in the welcome screen below the voltage, or the wellcome.
Please add option in menu, to edit it!
Very useful to put our Name, hamradio license number...

ErikS-web, 2023-11-24:
What modulation ?

This radio can only TX in FM.

MachoEnLaNoche, 2023-11-24:
> You can Adjust the welcome screen with Chirp [image]

Thanks for the info.
But it will better if it can be modifided from menu.
Thanks for your work!

ErikS-web, 2023-11-26:
On the PC with Chirp it takes less than 90 seconds work.
And after that I never changed it again. 

So why waste memory-space on the modify-function; the radio only has 60kB. 
That choice keeps coming back.

ErikS-web, 2023-11-26:
change 34. `POnMsg` - power on message, to value MESSAGE 

and use Chirp to set the Settings. Just like the above show lines.

And your screen shows

[image]

MachoEnLaNoche, 2023-11-29:
[https://youtu.be/wcS0kTk3DUc](url)

ErikS-web, 2023-11-29:
Have FUN with your HAM-hobby 
.
[image]

closed 2023-12-02 by egzumer