egzumer/uv-k5-firmware-custom #142: Tx on Air Band

Giagio89, 2023-11-24:
Hello! Thanks a lot for you wonderfull work. I have an aviation license so I can broadcast on aviation bands. What are the parameters that need to be modified to be able to speak on those frequencies? Obviously at my own risk

ErikS-web, 2023-11-24:
What modulation ?

Remender that this radio only does FM wih TX.

Giagio89, 2023-11-24:
Thx for aswer.
On AM.
 With this firmware (https://whosmatt.github.io/uvmod/) I could transmit.

ErikS-web, 2023-11-24:
Only RX, no TX.

f5gkw, 2023-11-24:
IMHO, don't use this radio on airband despite you have a licence to do.
It is not only your risk, it is also risks for others users as this radio
is full of harmonics.

ErikS-web, 2023-11-24:
Suggest to read this.... https://en.wikipedia.org/wiki/Airband

egzumer, 2023-11-24:
Harmonics are not a problem on those frequencies. The lack of AM modulation is. If you want to transmit on airband buy a proper airband radio. Airband communication is a serious thing, I'm not going to help you to spam on that band. Sorry.

closed 2023-11-24 by egzumer