egzumer/uv-k5-firmware-custom #148: In spectrum mode, after deleting the decimal point, it is not possible to input the decimal point again

wu58430, 2023-11-26:
In spectrum mode, press 5 to input the frequency. If the decimal point is removed, it will no longer be possible to input the decimal point. Exiting the input state and then re-entering can solve this problem. I tried using the following modifications to solve this problem（in spectrum.c ）:

[image]

So far, it has been working very well!
Thank you for your work!!!

ErikS-web, 2023-11-26:
I can reproduce this.

wu58430, 2023-11-26:
> I can reproduce this.

> I can reproduce this.

Yes, your execution ability is very strong. This is just a small issue. Thank you very much for your work

closed 2023-11-26 by egzumer