egzumer/uv-k5-firmware-custom #161: region `FLASH' overflowed by 64 bytes

Andandaran, 2023-11-29:
Downloaded v0.19.zip from releases and compiling (linking) gives this error:

arm-none-eabi/bin/ld: firmware section `.text' will not fit in region `FLASH'
arm-none-eabi/bin/ld: region `FLASH' overflowed by 64 bytes
collect2: error: ld returned 1 exit status
make: *** [Makefile:429: firmware] Error 1

using ubuntu 22.04 aarch64 with toolchain 10.3.1:

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)

I compiled v0.17 with same toolchain and I get the exact firmware.bin as yours (same md5)

I modified Makefile to:
- add /my_toolchain_path/bin/ to arm-none-eabi-*
- rename your target clean to clean-all
- create a target clean with only: $(RM) $(call FixPath, $(TARGET) $(OBJS) $(DEPS))
but those modifications have nothing to do with the error.

With toolchain 13.2.1 gives:
arm-none-eabi/bin/ld: region `FLASH' overflowed by 208 bytes

$arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arm GNU Toolchain 13.2.rel1 (Build arm-13.7)) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Andandaran, 2023-11-30:
> The "same md5" is an id of current git commit.

I mean md5sum of your firmware.bin and the resulting firmware.bin after compilation, not the md5 of the git commit.

Anyway I'll investigate more.

egzumer, 2023-11-30:
That is why I mentioned it, there is no md5 checksum enywhere.

Andandaran, 2023-11-30:
> That is why I mentioned it, there is no md5 checksum enywhere.

Right, you should provide the md5 of your binary firmware to give the possibility of verifying.

I did:
$ md5sum egzumer_v0.17.packed.bin 
6e9f0f6e22390448558633e18088b00c  egzumer_v0.17.packed.bin

and I did the same with the firmware.packed.bin compiled by me:
$ md5sum firmware.packed.bin 
6e9f0f6e22390448558633e18088b00c  firmware.packed.bin

This shows that the firmware.packed.bin I compiled is exactly the same as your egzumer_v0.17.packed.bin

Don't worry, as I say, I'll investigate what happens.

egzumer, 2023-11-30:
Ok, now I get what you mean.
My releases are auto built for some time now. They are built using github actions. I modified the action to use docker built method. That is why you get the same bin as v17 but not with v19.

https://github.com/egzumer/uv-k5-firmware-custom/actions/runs/6809915802
https://github.com/egzumer/uv-k5-firmware-custom/actions/runs/7026939824