egzumer/uv-k5-firmware-custom #170: When naming channels in the menu item 18, the frequency will change

losehu, 2023-12-02:
https://github.com/egzumer/uv-k5-firmware-custom/assets/52628935/c1739798-f56f-4d06-b562-837fe326d736

The frequency changes to the frequency of the main interface

losehu, 2023-12-02:
I think the purpose of this menu should only be to change the name, rather than changing the frequency as well

ErikS-web, 2023-12-02:
Compliments: Nice video to reproduce this issue.

Was a usb-powercable or programcable connected ?

closed 2023-12-02 by egzumer

kamilsss655, 2023-12-22:
@losehu, @egzumer I think the issue illustrated here is that some menu items should only be allowed for **memory channels** like:
```
	MENU_S_ADD1,
	MENU_S_ADD2,
	MENU_DEL_CH,
	MENU_MEM_NAME,
	MENU_1_CALL
```
and some should only be allowed for **vfo channels**:
```
	MENU_MEM_CH
```

There was no guard check in place, so I implemented a little function that checks if given setting is allowed, and if not the display shows `NOT ALLOWED` to prevent accidents like you illustrated above.

Here is the commit with the fix: https://github.com/kamilsss655/uv-k5-firmware-custom/pull/9/commits/d01952ed6cee230eac046f3ac12c4a443cb97521

commit d944ff4a (2023-12-02): Fix #170: ChName saves a name with wrong settings when used in frequency mode