egzumer/uv-k5-firmware-custom #174: S-meter in v.19 is incorrect for the VHF/UHF band

UW5EOT, 2023-12-02:
The S-meter again corresponds to the short wave (HF) range up to 30 MHz, but the main function of this radio is to work on the ultra-short wave (VHF/UHF) range of 145 MHz and 430 MHz, this is clearly higher than 30 MHz, which means the table should be for ultra-short waves (VHF/UHF), as was implemented in firmware v.17.

from http://hamwaves.com/decibel/doc/iaru.region.1.s-meter.pdf :
"The IARU Region 1 recommends the use of the "S"-system for signal strength reporting on the amateur bands, based on the following standards:
(a) One S-point corresponds to a level difference of 6dB.
(b) On the bands below 30 MHz a meter deviation of S-9 correspond to an available power of a CW signal generator connected to the receiver input terminals, of -73dBm.
**(c) On the bands above 30 MHz this power shall be -93dBm"**

from https://wd0gof.files.wordpress.com/2019/01/s-meter-standard-1.pdf :
"The International Amateur Radio Union (IARU) Region 1 agreed on a technical recommendation for S Meter calibration for HF and VHF/UHF transceivers in 1981. IARU Region 1 Technical Recommendation R.1 defines S9 for the HF bands to be a receiver input power of -73 dBm. This is a level of 50 microvolts at the receiver's antenna input assuming the input impedance of the receiver is 50 ohms.
**For VHF bands the recommendation defines S9 to be a receiver input power of -93 dBm.**"
[image]

egzumer, 2023-12-02:
See [#53](https://github.com/egzumer/uv-k5-firmware-custom/issues/53#issuecomment-1827981027)

My knowledge is to poor. I can only add that there is an [AGC](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Development-notes#how-it-works) active. It seems like the RSSI is a value at the output of all of the signal amplifiers. I suspect all gains should be added to the value of dBm calculated from RSSI.

egzumer, 2023-12-02:
> In the radio there is a Pre-amplifier between the antenna and the chip so the gain of that has to be taken into account. That was what the recent calibration was doing.

From what I've learned in last 2 days this gain is being adjusted depending on the strength of the input signal. These values can be any of:  [-7dB, -24dB, -43dB, -58dB, -79dB](https://github.com/egzumer/uv-k5-firmware-custom/blob/main/driver/bk4819.c#L307)
This is not taken into account right now.

UW5EOT, 2023-12-02:
> The RSSI values are now pretty close to being correct. The RSSI reading from the chip is for the input signal to the chip. In the radio there is a Pre-amplifier between the antenna and the chip so the gain of that has to be taken into account. That was what the recent calibration was doing.
> 
> This same discussion about S-Meters has been had many times before, on many platforms, about many different radios! The IARU table is a **recommended** calibration for SSB and CW radios. It also takes no account of the receiver IF bandwidth, which has a significant effect on the noise floor. Whilst the suggested figures may be reasonable for good SSB or CW transceivers with narrow filters they do not relate to FM or AM radio such as the UV-K5 which has much wider filters and average sensitivity. Wider filters have the effect of increasing the noise floor of the receiver.
> 
> The S unit has no absolute calibration. It was always intended as a relative signal report with each S level defined as
> 
> 0 Signals not detectable 1 Faint—signals barely perceptible 2 Very weak signals 3 Weak signals 4 Fair signals 5 Fairly good signals 6 Good signals 7 Moderately strong signals 8 Strong signals 9 Extremely strong signal
> 
> Based on this you would always expect an S meter reading of 0 when the radio is receiving no signal,
> 
> A typical handheld VHF/UHF FM receiver such as the UV-K5 has a noise floor of about -130dBm. So it makes sense to set the S meter to read zero at this level.
> 
> If it was set as per the IARU recommendation then the S meter would indicate around S3 all the time. Making S0 - S2 unusable.
> 
> Colin G4EML

What you wrote is only true for primitive walkie-talkies with a graphic display of 9 sections, but I am writing that we have the ability in this walkie-talkie to display numbers that correspond to the standard. And the VHF/UHF standard suggests -93dB for the S9, not -73dB as was done in the V19. In v17 this is done more correctly!!! And if you want a display from zero, please, it’s easy to calculate, but from standard conditions for VHF/UHF!

[image]

egzumer, 2023-12-02:
> We speak about a 15 bucks talkie

Doesn't matter how much it costs. I don't mind adding correct calibration if you guys come to some conclusion. I don't know who to listen to.

g4eml, 2023-12-02:
I agree it is not worth discussing much further for a cheap radio. 
My last comments follow, then I will leave it up to egzumer to decide what you want to do.

The varying gain of the preamplifier across the bands **is** now being compensated for. That was what the last calibration did. 

The only definitive measurement of signal level is the dBm reading. After the last calibration this is now as accurate as can be expected for a radio of this type and does not need changing. The signal level at the antenna socket corresponds very well with the displayed dBm value. 
If you want accurate absolute signal level measurements then you already have them, just use the dBm values. 

UW5EOT seems to be concerned about the mapping of the S scale to the dBm values.  This can easily be adjusted by changing the value of s0_dbm in ui/main. 
 
The way I see it there are basically four choices none of which are 'correct' but are more down to user preferences.

1. Leave the S0 level at -130 so that the radio indicates S0 with no signal.  S9 will then be at about -76dBm.
2. Adjust the S0 level to -147 so that S9 = -93dBm.   At the current 6dB per S point It will then never indicate lower than S3.  
3. Make the S0 Level a user adjustable setting in the menus. People can then adjust the radio however they like. 
4. Adjust the S0 Level to -120 and change to 3dB per S point so that S9 = -93dBm. 

Interestingly I have just measured the Icom IC905 radio on FM and they have taken option 4.  S9 = -93dBm and 3dB per S Point. 

Colin G4EML

UW5EOT, 2023-12-02:
> I agree it is not worth discussing much further for a cheap radio. My last comments follow, then I will leave it up to egzumer to decide what you want to do.
>
> The varying gain of the preamplifier [...]

I have nothing against calibrating the received signal in dB, only in accordance with the dB points S1...S2...S9... taking into account the VHF/UHF standard where S9=-93dB. When assessing communications, we usually call the values ​​S... and not the values ​​in dB. It’s easier for us to perceive the meaning by ear, for example, “I’m receiving you at 9+20” than “I’m receiving you at -73dB”, and in firmware v19, instead of -73dB, a much lower value is displayed than 9+20, which causes misunderstandings under equal communication conditions))

gartnl, 2023-12-08:
> 4. Adjust the S0 Level to -120 and change to 3dB per S point so that S9 = -93dBm.
> 
> Interestingly I have just measured the Icom IC905 radio on FM and they have taken option 4. S9 = -93dBm and 3dB per S Point.
> 
> Colin G4EML

That makes sense to me and seems a good compromise. 

As you pointed out there is much discussion on the subject of calibrating the S-meter. Modern receivers ditched the S-meter and have gone for either RSSI ( reception indication based on the receivers abillity ) or uV and SNR ( many Tecsun and derivatives, WiFi stations etc ). Only HAM's seem to stick to the S-meter.
Now I'm a HAM but don't see the point of displaying S3 to indicate the noisefloor of my receiver. And if, like many of us, you live in an urban enviroment with a manmade noise of  ~ -100 dBm you would read S6-S7. That renders the bargraph useless.

For the quality of the received signal SNR is far more important than the absolute signal strength. If the noisefloor is -100 and the signal is -93 that still means it is barely readable, even if it is S9. Pre 1981 that would have been S1. The IARU scale was usable way back in the past century somewhere in the countryside with a noisefloor of -150 but simply isn't usable in modern times.

Using method 4 as the source for the bargraph gives a better visual indication of reception quality, even if it is not mathematical correct. The numerical S-value could still be the 'true' value as to IARU.

I would even go so far as to ditch the numerical display of the S-value and replace it with SNR. But that would require a lot of additional prgramming. The spectrum uses this to adjust the lowest level to the actual noisefloor that is received, and scale the displayed value of a signal to that. It requires monitoring the noiselevel when no signal is received, and stop monitoring the moment the squelch value is exceeded. 

@egzumer I apologise for this post. I do not mean to make live difficult for you. It is just a remark on the oldfashioned status quo when the S-meter is concerned. Disregard it if you want.

wmarkow, 2023-12-08:
```
And the VHF/UHF standard suggests -93dB for the S9
...
When assessing communications, we usually call the values ​​S... and not the values ​​in dB. It’s easier for us to perceive the meaning by ear, for example, “I’m receiving you at 9+20” than “I’m receiving you at -73dB”
```
I totally understand this point of view. With this standardized levels of S0...S9 it is easy to compare between two people who speak to each other. There is no misunderstandings there. Otherwise its like talking about temperature: one thinks in Celsius degrees and the other in Fahrenheit.

```
Adjust the S0 level to -147 so that S9 = -93dBm. At the current 6dB per S point It will then never indicate lower than S3.
```

This means that the device reception sensitivity is maybe not so bad but not excellent as well, so it is not possible to exchange the communication with better devices and points to the direction that the user must be aware of the capabilities of his radio.

```
Now I'm a HAM but don't see the point of displaying S3 to indicate the noisefloor of my receiver. And if, like many of us, you live in an urban enviroment with a manmade noise of ~ -100 dBm you would read S6-S7. That renders the bargraph useless.

For the quality of the received signal SNR is far more important than the absolute signal strength. If the noisefloor is -100 and the signal is -93 that still means it is barely readable, even if it is S9. Pre 1981 that would have been S1. The IARU scale was usable way back in the past century somewhere in the countryside with a noisefloor of -150 but simply isn't usable in modern times.
```
Interesting piece of information. That could be a base of explanations about the levels of S-meter. However I think that noisefloor is also related to the sensitivity of the device itself, which is an output of the audio receiver design and the quality of the electronic components being used. UV-K5 is a low cost device so I assume that its parameters are not the top one. If you are located in the urban environment then the noise floor could be around -130dBm or even worse. When you go out to the countryside then the reception could be better and S-meter will show better results (towards to S0).

A few thoughts:
 * show received signal strength in dBm (as an absolute scale) but it should be well calibrated, just like it was done by @g4eml in #53
 * S-meter is a nice feature because for some people it's much easier to read the signal strength from the bargraph
 * some users argue what should be the level of S0 (or S9). Maybe make it configurable? Like it would be possible to enter the dBm level of S9. Then everybody could configure the radio according to its needs. Now I see that it requires some basic knowledge of waves propagation, but this has been explained and pointed in this issue and some other as well.

egzumer, 2023-12-08:
Thanks for all the input.
For me the best setting is what it is now as it has a wider range and it shows the actual signal strength and not a noise level.
The 3dB step is too narrow in my opinion. The level would jump with every slight change in signal strength.

I'll add the S0 and S9 configuration option, but this would be configurable only with external software, or possibly with a web browser. Need some time to implement that.

gartnl, 2023-12-09:
Thank you for the time and effort your are putting in to it!

wmarkow, 2023-12-09:
```
I'll add the S0 and S9 configuration option...
```
@egzumer , I'm not quite convinced if adding both S0 and S9 to the settings is a good idea, because giving the end user the option to configure both may be error-prone as the end user may enter inconsistent data (i.e. S0=-150 S9=-200). Also additional validation on the device should be done to check if the provided values have sense.

I propose to add either S0 or S9 to the configuration, then everybody can set it according to the needs. The span between two neighboring S-levels (in a range between S0 and S9) is always 6dB, so providing the level of S0 (or S9) i good enough to recalculate absolute dBm into its S-level value.

For me personally setting S0 is more human friendly as I can imagine this as a "noise floor" level or "beginning of the scale".

gartnl, 2023-12-09:
> egzumer , I'm not quite convinced if adding both S0 and S9 to the settings is a good idea, because giving the end user the option to configure both may be error-prone as the end user may enter [...]

If the values are set using an additional program as egzumer  intends that would make checking easy, and not consume space in the firmware. The stepsize in dB cannot be 6 dB: if  you leave S9 at -93 and change S0 to a higher value like -120. It would become (-93 - -120 ) / 9 = 3.

UW5EOT, 2023-12-09:
> Thanks for all the input. For me the best setting is what it is now as it has a wider range and it shows the actual signal strength and not a noise level. The 3dB step is too narrow in my opinion. [...]

That would be amazing. Everyone could configure the S-meter as they consider convenient or correct.
Thank you very much for your work, we enjoy using v.17 and look forward to the changes.

TNT777, 2023-12-10:
v.17 is very light, any small signal is very strong.
v.18 is the most real, it matches my Yaesu FT-991a.
v.19 is very heavy, rarely does a station exceed +9.
I did several tests, v.18 is suitable.

mentanah, 2023-12-13:
just flashed v.20 and noticed, that the s-meter always goes to maximum value, even on very poor receptions. do i have to calibrate it now with chirp?

egzumer, 2023-12-13:
> just flashed v.20 and noticed, that the s-meter always goes to maximum value, even on very poor receptions. do i have to calibrate it now with chirp?

Fixed #252

closed 2023-12-13 by egzumer