egzumer/uv-k5-firmware-custom #179: Alarm can transmit everywhere

ea7hjb, 2023-12-03:
Hi,

Even if TX is not fully unlocked, ALARM can do TX everywhere (including airband) if menu option 45 is set to "TONE".

Just for the info, 1750 Hz does show "TX DISABLE" as expected, so that function is OK.

Thank you for your effort and time and best regards.

ea7hjb, 2023-12-03:
Hello. If option is set to "SITE" it will only play on the speaker. If set to "TONE" it will do both TX and Speaker (switching between the two option each cycle).

If set to "TONE", it will not obey the TX table and will emit everywhere.

I remember I posted this exact same bug on 1of11 GitHub, and she fixed it, in case you can check that there and save yourself some time :)

ea7hjb, 2023-12-03:
Found it!:
https://github.com/OneOfEleven/uv-k5-firmware-custom/commit/b385b53af0b09ce1bf4928f46fede255e99b2dbf

In case it helps.

MachoEnLaNoche, 2023-12-04:
How activate alarm?

Arikshten, 2023-12-06:
This alarm function (IMVHO) is one to be removed and its memory used for better functions. 
I have never found a use case where it is appropriate/sensible/helpful for this to even exist. Although it might be simpler to fix than remove.

egzumer, 2023-12-07:
Fixed in cd032c39d28c1199cd4cf2921480d672f186f373

closed 2023-12-07 by egzumer