egzumer/uv-k5-firmware-custom #189: [BUG] - PTT ID won't save user's choice after power off

andreadeangelis2002, 2023-12-04:
Hello, I would report an issue about the PTT ID setting.

When configuring the PTT ID in the radio and saving the settings, the PTT ID is correctly set and functional. However, upon turning off and then restarting the radio, the PTT ID reverts to OFF instead of retaining the user-configured value.

The PTT ID setting should persist through power cycles, maintaining the user-configured value even after the radio is turned off and on again.

Thank you for the wonderful project!

egzumer, 2023-12-04:
PTT ID is per channel setting, you are trying to set it in a channel mode without saving channel settings

andreadeangelis2002, 2023-12-05:
Thank you for the clarification. I changed the title to this issue marking it as question instead of bug. Cheers!

closed 2023-12-05 by andreadeangelis2002