egzumer/uv-k5-firmware-custom #193: DTMF decoding failed

losehu, 2023-12-05:
I have two K5 machines, and the setting is to send the online code and enable DTMF decoding
When machine A sends DTMF, another machine can successfully decode it, but if machine B immediately sends DTMF, machine A cannot decode it, or if A sends DTMF again, B cannot continue decoding
That is to say, the machine needs to wait for a period of time after receiving/transmitting DTMF before it can continue to receive DTMF
When I use your firmware to continuously transmit DTMF to 1o11 firmware, 1o11 can receive it every time
But when I use your firmware to continuously transmit DTMF to another machine using your firmware, I cannot accept it
This video shows anomalies, hoping to help you understand the problem I am encountering

Uploading 12月6日(1).mp4…
The initial few short interval transmissions in the video cannot be decoded, but after a long distance, it can be decoded normally
Thank you very much for your work!

losehu, 2023-12-05:
https://github.com/egzumer/uv-k5-firmware-custom/assets/52628935/aadbd97b-9e85-4cf9-82a3-784f55ee052e

losehu, 2023-12-05:
I found through debugging that when an exception occurs, the loop in the "static void CheckRadioInterrupts (void)" function in the "app/app. c" file will immediately end during transmit/receive, and during normal operation, this loop will execute multiple times
This is the loop：
 while (BK4819_ReadRegister(BK4819_REG_0C) & 1u)

losehu, 2023-12-06:
> Thanks for signal / issue with the video. Are both equipt with version 0.19 ?

Yes, great firmware

Arikshten, 2023-12-06:
I might have seen something related/similar. 
I have been playing with DTMF calling and one of my sets rarely seems to recognise when it is being called. The display shows the DTMF it is receiving as missing the first digit most of the time (as I see in the above video). Both devices are on V0.19 (fantastic firmware and very much appreciating your work) with the same settings. Strangely it is only ever that specific unit and the other one never has the issue (thus far in my limited testing). I have also set the Squelch the same on both devices. 

I wonder if there are differing calibrations in some sets and if a tiny delay before the first DTMF signal would help. 
For what it is worth I had played a little with this in the manufacturers firmware and was also having similar issues but unable to see much of what was going on as it wont display the DTMF received.

Adding this note in case the information is of help and in case I can help with any testing etc. to fix this.

losehu, 2023-12-06:
> I might have seen something related/similar. I have been playing with DTMF calling and one of my sets rarely seems to recognise when it is being called. The display shows the DTMF it is receiving as [...]

What confuses me is that when using egzumer's 0.19 firmware to send DTMF, 1o11 can decode very well every time. Conversely, using 0.19 firmware to decode cannot achieve 100% success
In fact, this issue was discovered by me during the process of porting the MDC1200 feature of 1o11 to the egzumer firmware. The firmware I modified has a probability of successfully decoding the MDC1200 ID, and sometimes decoding may fail. However, the emitted data can be 100% decoded by Motorola. During debugging, I discovered the loop I mentioned earlier, and it was the early abnormal exit of that loop that caused occasional decoding failures, And this exception will not occur in the 1o11 firmware.
So, I speculate whether DTMF will also have the same issue in the original 0.19 firmware. After my testing, unfortunately, the original 0.19 firmware also has issues
I don't think the issue of DTMF decoding failure is caused by minor differences in the device, as the two 1o11 firmware can work well together

JuantAldea, 2023-12-23:
> I might have seen something related/similar. I have been playing with DTMF calling and one of my sets rarely seems to recognise when it is being called. The display shows the DTMF it is receiving as [...]

I Have experienced the "first character" issue as well.

Arikshten, 2023-12-26:
I have continued tested with egzumer's v0.21 version of the firmware and I have not seen the issue again at all.  
Everything looks to be working well for me now. Happy to test in any way if that is of help.

edyasia, 2024-01-13:
Hello, I know this is not the right medium but I have a question. Do you any firmware in English that supports MDC1200, I use the 1o11 one but I want to see if there is something better that has this function. Thanks in advance

losehu, 2024-01-13:
> 您好，我知道这不是正确的媒介，但我有一个问题。 有没有支持MDC1200的英文固件，我用的是1o11，但我想看看是否有更好的东西有这个功能。 提前致谢

> 您好，我知道这不是正确的媒介，但我有一个问题。 有没有支持MDC1200的英文固件，我用的是1o11，但我想看看是否有更好的东西有这个功能。 提前致谢

sorry, my firmware only build for chinese language.
if u wanna english ,u can try to edit chinese.h ,replace everything to english words, and change it's size 
it should be works after rebuild

commit f8ff71aa (2024-01-13): Fix #193: DTMF interrupt temporarily disabled after TX causing DTMF tones lost
closed 2024-01-13 by egzumer