egzumer/uv-k5-firmware-custom #194: AM-Fix Broken

g4eml, 2023-12-05:
Commit 86438ef 'Improve Am-Fix' seems to have broken something. 

In VFO mode, after receiving a strong AM signal a following weak signal is unreadable and causes a rapid pulsing of the S-Meter between zero and maximum.   It looks like the AM-Fix algorithm is rapidly oscillating between max and min gain. 

Reverting this commit fixes the problem.  I have not go the time to try to track it down any further at the moment. 

Colin G4EML

ErikS-web, 2023-12-05:
Thanks for signal / issue

egzumer, 2023-12-05:
https://github.com/egzumer/uv-k5-firmware-custom/actions/runs/7107247363

At the bottom there is a firmware to download, or build your own from devel branch. It has am-fix disabled and AGC is reconfigured. I tested with an aviation handheld, transmitting from 10m away the audio is clear on the receiving end. This needs to be tested for different strength signals. Will appreciate feedback.

g4eml, 2023-12-06:
I have tried the firmware and it does perform better that the original, but not anywhere near as good as AM-Fix. 
The severe distortion is fixed but the limited number of gain levels is very noticeable. On slowly increasing RF signal strength the audio starts off quiet, gets gradually louder then suddenly switches to quiet again at each AGC switching point. Also around some switching points it seems to oscillate between the two gain settings resulting in a loud buzzing on the audio. 

AM-Fix has a much finer control of the gain setting. It does still step as the signal level is increased but the change in audio level is much less variable. 

Of course on a conventional AM receiver the AGC is analogue and is constantly adjusting the gain for best audio level. As it stands AM-Fix is the closest to that behaviour. 

Colin G4EML

egzumer, 2023-12-07:
Well it is not fixed, I don't have any AM transmissions so I didn't test it well enough. Now I tested with any non AM signal and I see the issue. Working on that.

commit 1857e345 (2023-12-07): FIX #194: Spectrum going crazy when AM-fix is active
commit c6975968 (2023-12-07): FIX #194: Spectrum going crazy when AM-fix is active