egzumer/uv-k5-firmware-custom #199: Filter weather alerts by using SAME codes?

dpoisson2017, 2023-12-06:
Is it possible to have the weather alerts filtered by SAME codes (up to 3 or 5)?
Weather stations broadcast on a vast territory and alerts are often only for a specific county.

From https://www.weather.gov/iln/nwr_same:
At the beginning of the product transmission, 3 bursts of digital static are heard (which contains 6 digit SAME code, amongst other things)
Followed by 10 second 1050hz alarm tone
Followed by the actual audio message.

Has anyone ever looked into this? Would it even be feasable...I have doubts because of the "digital static" part possibly couldn't be demodulated? To my knowledge, this would make the Quansheng radios the only ham radios to have this functionality.

ErikS-web, 2023-12-06:
NOAA is standard NOT enabled in this firmware.

There is no use for it in Europa.

See [README.md](https://github.com/egzumer/uv-k5-firmware-custom/blob/main/README.md)
ENABLE_NOAA                   := 0       everything NOAA (only of any use in the USA)

egzumer, 2023-12-06:
From the datasheet it looks like the RF chip supports the NOAA SAME code decoding, but I'm from Europe where there is no NOAA system, I have no motivation to work on that.