egzumer/uv-k5-firmware-custom #204: DTMF whilst receive airband?

kittenarmy, 2023-12-07:
Using v19. Not sure if this is a bug but I didn't notice it before in other firmware. I get this string appearing sometimes whilst receiving, not pressing any buttons or hearing any DTMF tones.

[image]

[image]

Video clip:
https://imgur.com/a/stgPudN

egzumer, 2023-12-07:
It is how it works. It takes noise for DTMF codes, this is how the RF chip works. The live DTMF isn't really useful and probably should be deleted from default build. It can also be disabled in the menu.

kittenarmy, 2023-12-07:
Ahh thank you!

closed 2023-12-07 by kittenarmy