egzumer/uv-k5-firmware-custom #216: Feature request: on-device editing for "ANI ID," "UP CODE," and "DOWN CODE" settings

andreadeangelis2002, 2023-12-08:
Hello, I would like to propose the implementation of a new feature in the firmware that allows users to modify the "ANI ID," "UP CODE," and "DOWN CODE" settings directly from the radio's keypad. This enhancement would eliminate the need for users to connect the radio to a PC every time they want to make adjustments to these settings, like has been done for the single channel name. That would enhance the convenience and accessibility of the radio configuration, especially in situations where connecting to a PC is not practical or feasible.

I would like to take this opportunity to express my appreciation for the excellent work done on the firmware thus far!

ErikS-web, 2023-12-08:
Did you read page 12 of the original manual. Chapter 6.11 DTMF calling ?

andreadeangelis2002, 2023-12-08:
Thanks for the reply.
Of course I did, but sadly there is only the option to manual dial a DTMF call. What I meant was about editing the radio ANI ID and down/up code from the keyboard, without the need to do it via PC. Just like has been done with the channel name option. Thanks for considering this topic.

JollyH0789, 2023-12-11:
Great idea, I would like this feature too

BruMcD, 2024-02-12:
They are normally not available in the set's menu as in commercial use changing these 'on the set'  provides a clever user an opportunity to transmit or receive 'as someone else' and /or to  'dodge' Kill /Stun commands.

For hobby use it may be neutral but I can't immediately see why you would want to do it 'on the fly' unless trying codes out for some strange purpose