egzumer/uv-k5-firmware-custom #226: no reception on the 2M band

MPHI77, 2023-12-10:
I can no longer get any radio in the 2M band. The meter shows full reception, but there is nothing except noise. Unfortunately the problem only occurs with your mod. testet with 017,018,019.

gartnl, 2023-12-10:
Did you make sure demodulation is set to FM? See menu 13.

ErikS-web, 2023-12-10:
do you suggest that with firmware 16 it is ok ?

pse check...

13 `Demodu` - demodulator mode, default is **FM**, AM/USB can be used for listening only

MPHI77, 2023-12-10:
Ok, I have to apologize. It was actually my fault. I had also tested the demodulation on FM, but didn't think about it when I turn the radio off and on again that the demodulation would be back on AM. I have now set the frequencies from AM to FM in Chirp and now everything is fine again.

closed 2023-12-11 by egzumer