egzumer/uv-k5-firmware-custom #227: padlock icon looks like a candel. Change it for a key

MachoEnLaNoche, 2023-12-10:
It is like a candel. In the top center of sreen, when is locked, please put a big padlock, a key or just "LOCKED".
NOW is confuse, it is like an arrow to the led torch...

PLEASE INCREASE THE PRESETS OF FM RADIO TO 100
PLEASE. 100 or unlimited, when scan automatic

Hunterfnet, 2023-12-10:
Totally correct and for same price and time can change the list selected chopsticks for L1 or L2 or similar for a better view , the radio FM broadcast i leave for put on memory the half part of broadcast stations. It's a very good modded firmware with the little "things" and failures as all normal great things . Totally on same way "machomanonthenight"

ErikS-web, 2023-12-10:
Locked is BIG on the screen when you press a key that is locked.
This does not apply to all keys, such as PTT.

Based on your other wishes...
FM radio memory channels will not be expanded. Buy another radio than this one.

ErikS-web, 2023-12-10:
Your first line ... It is like a **candel**. In the top center of screen, when is locked.

Is an issue to investigate.

MachoEnLaNoche, 2023-12-10:
In original firmware the padlock is BIG AND WIDER.
Now the padlock looks like a candel a the top center of screen, pointing out the led torch. I little confuse.
Put a KEY, a Big padlock o a "LOCK" text.

Changing the fm broadcast memories is easy, but I am not programmer, but i have read the fm.c file... I think it is just change 20 for 100.

gartnl, 2023-12-10:
that's a lot of memory . It's is not just upping the number but find storage for it. 80X2 = 160 bytes eeprom. Plus additional coding.
Quite an effort for a HT that only has WFM as a bonus.

ErikS-web, 2023-12-10:
See #139 "_20 stations is plenty enough. I'm not willing to work on expanding that,_"

MachoEnLaNoche, 2023-12-10:
So, to save memory and leave the FM radio with a good image, you should: delete the MR function, leaving only VFO.  and that when you click on SCAN it will only allow you to use automatic manual scan up and down without saving in memory.  and pressing VFO/MR returns to VFO fm.  BUT WHAT WOULD REALLY BE GOOD IS: PRESS FM>ENTER FM - SHORT PRESS UP DOWN UP DOWN FREQUENCY, LONG PRESS UP DOWN SCAN UNTIL FOUND AND STOP, and on the main fm radio screen.  that says FM RADIO.  AND EXPAND THE BAND UP TO 65MHZ 108MHZ.  to cover the FM band of all countries, like the Baofeng walkie.

egzumer, 2023-12-10:
It doesn't look like you had any problem identifing it as a padlock. It was changed to look like that by 1o11. It is what it is, it doesn't bother me. I'm not changing that. Feel free to do your own modification it is not that hard.

I will add unlimited FM radio presets when somebody else invents unlimited memory mod, come back when that happens.

closed 2023-12-10 by egzumer