egzumer/uv-k5-firmware-custom #228: Scan in monitor mode

marianomd, 2023-12-10:
It would be nice to have an option to scan frequencies with open squelch. As far as I understand, it wouldn't be able to stop on it's own, the user would have to stop it manually. This would be nice to detect faint transmissions that fall under level 1 squelch.

Ps: I love your work and dedication, thanks.

ErikS-web, 2023-12-10:
Buy yourself an analog receiver.

I think an analog radio better suits your needs. Turn squelch open and tune the radiomanually.

gartnl, 2023-12-10:
Nearest thing is to set sql to 0, start scanning and use up/down to manually step throught the scangroup only.

marianomd, 2023-12-10:
Yes @gartnl that's what I do, but need pressing the up key all the time.

marianomd, 2023-12-10:
@ErikS-web If the hardware allows it, why should I buy another radio?

ErikS-web, 2023-12-11:
Because the radio does not have such a nice rotary knob as an analogue radio.
That is especially useful if you run across the frequency band with the squelch open.

gartnl, 2023-12-11:
But what use would that be? You don't detect every transmission because stepping through the range takes time. You'd have to sit next to it hours on end. And since it does not stop on a signal the moment you hear it it is gone again.
The best thing for you is to get an sdr and use computersoftware to store any frequency where a signal is detected. Or even record the IQ and look at it later on. But don't ask the impossible from a simple handset.

closed 2023-12-11 by marianomd