egzumer/uv-k5-firmware-custom #230: expand fm radio mode frequency

QuickXand, 2023-12-11:
is there any possible to unlock the fm radio mode to 65Mhz? there's some radio station running under 78Mhz and the bk1080 datasheet say that support 65-108Mhz.

egzumer, 2023-12-11:
From a brief look on a band plan, Japan is the only country that has different frequencies for FM broadcast. It doesn't make sense to expand that for everyone. I think the best option would be to add a build option for that.

gartnl, 2023-12-11:
Even Japan starts at 76 , not 65. 76 is already covered. The lower 65 - 74 is only used in some eastern-european countries to be compatible with older equippment. But with a different channel spacing. Most of these are duplicated on 87-108 anyway as there is a shift to that range.

Hunterfnet, 2023-12-11:
I'm sure that for now you can do something setting down the FM settings with the UVMOD or maybe changing the channel frequencies why Chirp. On few time surely you can change a lot of things with a simply windows software.

QuickXand, 2023-12-12:
> From a brief look on a band plan, Japan is the only country that has different frequencies for FM broadcast. It doesn't make sense to expand that for everyone. I think the best option would be to add a build option for that.

it might cant be easily expand to 65Mhz...the datasheet says that 64-76Mhz is a stand-alone mode and need more work on driver
:(
[image]
im not sure if this feature is worth the cost...

QuickXand, 2023-12-12:
> I'm sure that for now you can do something setting down the FM settings with the UVMOD or maybe changing the channel frequencies why Chirp. On few time surely you can change a lot of things with a simply windows software.

it need more work on driver.The band register must change if u want tune any frequency on 64-76M

ErikS-web, 2023-12-12:
Can you explian more...

What kind of radio-station ? 
What country and frequency ?

egzumer, 2023-12-12:
Pinning resource
https://en.wikipedia.org/wiki/FM_broadcast_band

gartnl, 2023-12-13:
> Can you explian more...
> 
> What kind of radio-station ? What country and frequency ?

Broadcast stations in East Europe. Search for OIRT FM to find out more. 
But they are almost completely phased out in favor of the 87.5-108 CCIR bandplan.
It may be technical challenging to enable this but I wonder if it is of any practical use.

gartnl, 2023-12-21:
@egzumer If you are still working on this I have an additional request: make the range selectable (CHIRP driver would be a place).
Combined with the autoscan feature it would be nice to start scanning at the actual lowest frequency for the region.
In the current firmware is it easy to change the lowerlimit ( settings.c ) and recompile, but if the additonal range is added that might be a problem.
As I read it upper and lower limit are hardcoded , but there is some space in the FM array ( 8 bytes padding ? ) to store this.

gartnl, 2023-12-30:
I did an experimental build with OIRT included and selectable regions . I seems to work, I do receive in the OIRT range. I don't have a generator to test broadband FM, but I do receive a mirror from a strong local station at 104 in this band on 65.2. 
I made it so that even the full 65-108 range could be possible, BK1080 is switched based on frequency. It is also possible to have mixed frequencies in memory locations.
So it can be done, but if it is of any use? It uses appr 75 bytes of precious memory.

gartnl, 2023-12-31:
[experimental_OIRT.zip](https://github.com/egzumer/uv-k5-firmware-custom/files/13800683/experimental_OIRT.zip)

I'm way behind egzumers firmware now and still on v0.20.1. But you could try the firmware in the zip to see if it works for you, I'm curious if it does. The zip also contains the modified CHIRP driver.
Also note: the internal 26 MHz xtal produces mirrors in the OIRT band for strong FM stations above 104 MHz.

gartnl, 2024-01-08:
As I wrote above, selecting the OIRT band was easy enough. But I'm running into problems with the BK1080 channels. 
OIRT uses 30 kHz channelspacing. But none of the datasheets I could find mentions this option. It only states 200/100/50. 
Is there anyone who owns a datasheet that has more info on this? Could the fourth option ( it is a 2 bits option ) be 30?
It would be strange for the chip to support OIRT but not the correct channelspancing ..

QuickXand, 2024-01-13:
> As I wrote above, selecting the OIRT band was easy enough. But I'm running into problems with the BK1080 channels. OIRT uses 30 kHz channelspacing. But none of the datasheets I could find mentions [...]
That's all and there's no more option.

[image]
The radio station running beyond the original limitation was closed so i cant test this build...for now.But I can sure it was good when receiver channel spacing is default.

gartnl, 2024-01-14:
> The radio station running beyond the original limitation was closed so i cant test this build...for now.But I can sure it was good >when receiver channel spacing is default.

Seems OIRT is still used by some local stations but with the 30 kHz channelspacing, but frequency lists are hard to find. Main stations moved to CCIR. 
The build with OIRT support was an experiment, but for my own curiousity.

egzumer, 2024-01-15:
I added band switching with `1 BAND` button. I tried to add step switching but it is complicated. All frequencies are saved with 1 decimal digit, so going down below 100kHz adds too many complications, and there is very little benefit to that. There is no 30kHz step, not even for SPACE=3.

crazypup, 2024-01-16:
I'm guessing that for most people using this firmware, their FM broadcast band starts at 88 MHz. So at very least I'd love to see the FM auto scan begin there by default.

gartnl, 2024-01-16:
> I added band switching with `1 BAND` button. I tried to add step switching but it is complicated. All frequencies are saved with 1 decimal digit, so going down below 100kHz adds too many complications, and there is very little benefit to that. There is no 30kHz step, not even for SPACE=3.

That is true. But there is also no benefit for 200 khz as these frequencies are allready covered with 100 kHz steps. 

The frequency is stored as a channel number. The actual frequency is calculated in the BK1080 as (lower_limit + channel*spacing).
Scanning is done manual, that is the channels are set in software and there is a check for a valid signal.
So if you omit OIRT only FM_LowerLimit and FM_UpperLimit need to be set. It doesn't matter that the selected band in the chip is 76-108. Scanning will start at FM_Lowerlimit and end at FM_Upperlimit. 
This is also true if you include OIRT. You would only have to switch bands if FM_Lowerlimit is below or above 76. 
This simplifies code.

The actual OIRT band starts at a different frequency than the BK1080. This also means the even if the chip supported 30 kHz spacing, the used method in the chip of (lower_limit + channel*spacing) would not result in the correct frequency.
All stations I found in the OIRT band use the 30 kHz spacing, and can't be tuned correctly.
All-in-all I dropped OIRT. I very much doubt that it is of any use when you can only tune incorrectly.
		

		{
			uint16_t SelectedFrequency;
			uint8_t  SelectedChannel;
			uint8_t  IsMrMode;
			// added region
			uint8_t  Region;
			//uint8_t  Padding[3]; // not needed when reading, contains no info and struct is packed
		} __attribute__((packed)) FM;
			
		EEPROM_ReadBuffer(0x0E88, &FM, 5); // no need to read more than 5 bytes

		// added bandlimits

		const uint16_t    Region_LowerLimit[] = {760,875,880,870,760};
		const uint16_t    Region_UpperLimit[] = {1080,950};	

		if (FM.Region > 4) FM.Region = 0; //sanity check

		gEeprom.FM_Region = FM.Region;
		gEeprom.FM_LowerLimit = Region_LowerLimit[FM.Region];
		gEeprom.FM_UpperLimit = Region_UpperLimit[FM.Region/4];