egzumer/uv-k5-firmware-custom #232: Write the font matrix into eeprom

losehu, 2023-12-11:
I found that there seems to be 50KB of unused space on the external EEPROM. If you first flash the firmware to write the font library to EEPROM, then flash in the normal firmware to read the font library from EEPROM, it can save more space for your code. I have tried this idea on my Chinese firmware, and it is currently working very well

egzumer, 2023-12-11:
50kB? As far as I know there is 8kB and most of it is already used.

losehu, 2023-12-11:
I have calculated the length mentioned inside, which is about 8k, so there should still be over 50k left. Perhaps there is a setting mentioned? Or did I make a calculation error?
https://github.com/amnemonic/Quansheng_UV-K5_Firmware/blob/main/docs/cfg_mem_map.md

egzumer, 2023-12-11:
The chip is 64kb not kB.

losehu, 2023-12-11:
Oh my god, I'm a bit foolish

closed 2023-12-11 by losehu

losehu, 2023-12-11:
You're right, I recalculated and there's still 300B left，Can larger frequency fonts be written in?
However, this requires burning the program once more, but at least it is a one-time fix

egzumer, 2023-12-11:
There is max 256kB of continous free space, there is also more then double of that in chunks overall. But it is not worth the hassle in my opinion.