egzumer/uv-k5-firmware-custom #234: FM-DW lock bug

gartnl, 2023-12-11:
This one is sometimes annoying. When listening to FM broadcast the radio still listens to the active VFO ( the passive is used for broadcast FM). That is what it should do. ( it should be noted this is not clear from the original manual, but during FM broadcast the DW option is not available )
If the active VFO's receives a signal that signal is given priority and activated. Again as it should be. 

But .. now the keyboard is locked! I have to wait for the transmission to end before I can exit or change anything. Only way out is turning off the radio.
Testing: set the active VFO to any test frequency. Than turn on broadcast reception. Now have a signal generator transmit on the test frequency. The keyboard is locked.

This is not! an egzumer bug but also present in the QS firmware. So I don't dare say that it is solvable.

egzumer, 2023-12-11:
The real bug is that dualwatch is not working while in FM radio mode.
The issue that you have is solved by simply pressing exit button, this turns off FM radio. The question is how one wants it to function, I agree this is not perfect, but otherwise you will have to solve another issue. What do you do with the FM radio mode when you start changing something, do you still bring it back?

gartnl, 2023-12-11:
Ah I see where I made a mistake. When you press the exit button there is no feedback so it looks like nothing happens.
When the HT goes out of FM mode to normal VFO when something is received there is no visual clue that the HT is actually still in FM mode. Pressing exit therefore seems to do nothing.

For me I would like 2 changes:
- visual indication that FM radio mode is active when the HT changes to normal VFO reception. 
- when in FM radio VFO mode , long press on up/down could start manual scan until next signal. 
Am I correct that this manual scan is missing in your firmware? I can only start auto scan. The QS firmware can start a manual scan with F-3. It will scan untill next signal. From there up/down continues manual scan.

I agree, it is difficult to decide what are usefull funtions.

egzumer, 2023-12-11:
Short press * is manual scan, longpress * is auto-scan.

commit 0c2fc618 (2023-12-11): Fix #234: Bring back FM radio status indicator, show dual watch inactive while in FM radio