egzumer/uv-k5-firmware-custom #241: new menue settings getting lost after programming channels

mentanah, 2023-12-12:
first of all, dear egzumer thanks a lot for this brilliant repository and the excellent development of the custom firmware.
i cant wait for release 0.2 since there are more than 50 new commits.

when i programm the channels and change the settings with original software, some new settings (battery type, backlight min max settings) are getting lost. Is it possible to preserve them?

kittenarmy, 2023-12-12:
Which PC software are you using? I suggest using [CHIRP](https://chirp.danplanet.com/projects/chirp/wiki/ChirpNextBuild) if you are not already. You can update channels and settings independently.

mentanah, 2023-12-12:
Thank you for this hint. The Radio ist already fully implemented in chirp?
I was using the Portable Radio CPS from Quansheng or Radtel.

egzumer, 2023-12-12:
Use CHIRP, don't open settings tab!!!
The driver specifically for this firmware is being developed.

ErikS-web, 2023-12-12:
quite simple... just do...

Radio - Download from Radio.
DO_NOT enter Tab Settings !
Modify Channels 
Radio - Upload to Radio

Enjoy.