egzumer/uv-k5-firmware-custom #248: Feature Request : Bring back power-on password / PIN

ssoespre, 2023-12-13:
firstly, appreciation to all whom has deeply involved developing this custom-firmware, very much exciting seeing an ordinary thing has been tweaked into something EXTRA-ordinary :)

i know this feature request has been abandoned because of memory limitation, so this is a proposal only.

from my point of view, we are all excited to tinker with all the settings, all the frequencies we put into our radio, so to make it even more secure and personal, i hope power on password / PIN is considered to be brought back into the firmware :)

thanks :)

egzumer, 2023-12-13:
Nope, not comming back in the official release. The build option are there so everybody could pick and choose. Build your own options if you want that feature.

closed 2023-12-13 by egzumer

ssoespre, 2023-12-13:
well noted, thanks though :)