egzumer/uv-k5-firmware-custom #250: Feature Request : A way to back-up the radio data

ssoespre, 2023-12-13:
i tried to read my custom V.0.19 firmware radio into original portable CPS for UV-K5, it seemed any frequencies that out of limit of original firmware cannot be read correctly, they will be changed into something that the CPS can accept.

so, i suggest either one of this :

1.  a custom CPS that can be used for reading and writing the content of custom firmware
2.  another way of backing up the content of the radio, so it may be restored someday if the radio got problems

once again, thanks for your passionate hard work :)

chouyu, 2023-12-13:
use chirp software

ErikS-web, 2023-12-13:
See our [online-manual in the Wiki](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Radio-operation)

and read [latest release notes](https://github.com/egzumer/uv-k5-firmware-custom/releases/tag/v0.20) for instruction to extended Chirp-driver for release 0.20