egzumer/uv-k5-firmware-custom #252: default values for s-parameters in v.20 wrong

mentanah, 2023-12-13:
i had to set the parameter: S0_S_meter_level_dbm from 1 to 130 with chirp, in order to get a propper working s-level indicator after flashing the v.20 firmware

gartnl, 2023-12-13:
Strange. I updated the firmware and default S0 was on 130 when I read it with CHIRP. But default for S9 on 76.

ErikS-web, 2023-12-13:
The values were wrong in the previous (test-)version of the driver and corrected yesterday in the new driver.

Pse use latest chirp-driver-version. See release notes

egzumer, 2023-12-13:
Not used parts of memory are filled with 0xFF. S0, S9 are located on such a space.
When it reads 0xFF it uses defaults of 130 and 76.
I don't see a bug in the code, maybe there is, but right now I would say that the 1 was written there by some other firmware.

egzumer, 2023-12-13:
> The values were wrong in the previous (test-)version of the driver and corrected yesterday in the new driver.
> 
> Pse use latest chirp-driver-version. See release notes

As far as I can tell it has nothing to do with that.

mentanah, 2023-12-13:
Maybe the memory wasnt 0xFF in my case due to former programming with original CPS Software from Quansheng. But the Problem occured before i ever used chirp anyway.

egzumer, 2023-12-13:
I'm going to add tighter constraints. It will be:

-200 < `S0` < -90
`S0` + 9 < `S1` < -50

Luca6605, 2023-12-16:
I had loaded the 0.20 but the s meter always showed s9. I fixed it with version 0.20.1.Thanks

mentanah, 2023-12-16:
> I had loaded the 0.20 but the s meter always showed s9. I fixed it with version 0.20.1.Thanks

I checked it definitely now. Programming with original CPS Software from Quansheng sets the S0 value to 1. V.20.1 then corrects it to 130. When you want another custom value there, you will have to correct it with chirp after every CPS programming.

egzumer, 2023-12-16:
@mentanah yes you are right, CPS uses that as a language setting, that was probably intended for switching menu languages on the radio but was never implemented by quansheng. Only CPS uses this memory address that is why I missed that. I will have to change that to some other address in the next release. It will be a little annoying for some, but it has to be done.

egzumer, 2023-12-16:
Actually I change my mind, CPS overwrites everything that is not standard quansheng settings, thus it should not be used with any other firmware than stock firmware. Reading invalid setting from that memory address is fixed, so everything should work flawlessly.

closed 2023-12-16 by egzumer