egzumer/uv-k5-firmware-custom #262: 3800mAh battery profile (enhancement)

aliatx2017, 2023-12-14:
Hello and thanks for your fantastic project and the time you put into it!

I know that some of us are getting uv-5r radios for its bigger 3800mAh battery. With a simple mod (break off center tab) this battery mounts and works in uv-k5(8)/k6 radios.

Would it be possible to add a 3800mAh battery profile to code/hidden menu?

Respectfully,
Alex

gartnl, 2023-12-14:
I wonder if it is really 3800 mAh. I've got 2 5R Plus's  and both have a stock 2200 mAh battery. 
The 3800 mAh batteries are sold as replacement, but those might be Chinese milliamps ..

egzumer, 2023-12-14:
Above answers are correct.
It is not about mAh, it's about cells chemistry. It might be the same as one of those that are already there.

These are discharge curve points (voltage, percentage):
https://github.com/egzumer/uv-k5-firmware-custom/blob/1a53586922880afbec678f3b855486918811057e/helper/battery.c#L54