egzumer/uv-k5-firmware-custom #268: Scanner and spectrum improvement ?

Robby69400, 2023-12-14:
Some idea I have:
does the frequency display each step slow down scan? Could we go faster not displaying frequency during scan but only when it stops?
My understanding is that the spectrum is much faster, no idea why but I would like you to consider some way to show a bigger frequency range, for eg showing the highest level in several steps. Tell me if I need to explain more.

mentanah, 2023-12-14:
For orientation I would like to see the approximate frequency while scanning. But every 10th frequency update should be enough. What do you think?

egzumer, 2023-12-14:
Display updates while scanner waits for the signal, the scanner wait time is 90ms, I don't think screen update takes more than that, and if it doesn't, then it doesn't matter.

Robby69400, 2023-12-14:
My main wondering is how spectrum looks much faster than scanning ? I like to use spectrum for that.  That's why I'm really curious to know if it would be possible to increase the spectrum bandwith even if it slows a bit.

egzumer, 2023-12-14:
Spectrum simply reads RSSI, it doesn't relay on RF chip logic of recognizing a valid signal (squelch etc.).
You already asked about more steps in spectrum, it is certainly doable, but I'm not planning on doing that right now. It already takes significant amount of time to do 128 channels scan, and with 25kHz bandwidth it should detect all signals within given range, that is over 3MHz.

egzumer, 2023-12-14:
duplicate #103

closed 2023-12-14 by egzumer

egzumer, 2023-12-14:
100kHz is a step, not bandwidth. 25k is the max