egzumer/uv-k5-firmware-custom #272: Incorrect backlight duration in version 0.20.1

F6EGK, 2023-12-14:
Several users and I, jump from version 0.19 to 0.20.1. Since then, backlight duration does not match with input value thanks to menu 36. It seems that backlight duration is now exactly twice input value, e.g. 10 s for 5 s input, 20 s for 10s input, etc.

Thanks for your efforts in pushing valuable firmwares to all the community.

Regards.

Jean-Roger / F6EGK

JuantAldea, 2023-12-14:
I'll take a look at it

mfarikan, 2023-12-14:
I tested it too and the backlight is set for 5 seconds and stays on for 10 seconds.

kittenarmy, 2023-12-14:
Seems some operation causes it to turn off on time (maybe the value isn't read always when exit setting menu after adjusting the time), but in my testing it often doubled as OP described. 0.20.1/rev1.6 UVK5(8).

JuantAldea, 2023-12-15:
There's a factor of 2 that was added by me. It is indeed a bug. I'm not sure how some of you are not reproducing this issue.

commit a99d2dd1 (2023-12-15): Fix #272: Backlight time doubled
closed 2023-12-15 by egzumer