egzumer/uv-k5-firmware-custom #275: Backlight reverts to full brightness

kittenarmy, 2023-12-15:
v0.20.1

Steps to reproduce:
Set BLMax to 5 (BLMin stay set 0), press M to confirm - backlight reverts to full brightness until press exit.
Press M and backlight is full brightness again.

Also triggered by:
Press M to open menu, scroll with up/down to BackLt - backlight reverts to full brightness until press exit.

JuantAldea, 2023-12-15:
I think it always lights at full brightness when pressing the M key, isn't in?

egzumer, 2023-12-15:
This is inherited from 1o11 and I left it. The idea is I guess to allow bringing back the backligth settings in a dark when it was set to off.

JuantAldea, 2023-12-15:
I have proposed solution of the "side effects" of this, although it is really cosmetic and it increases the binary size so I have mixed feelings about it.

gartnl, 2023-12-15:
I actually like the backlight at FULL when entering or traversing the menu, it's kind of a failsafe. Just imagine you are by accident setting MAX to 1 ( might happen if you think you are setting MIN ). Press exit twice or simply wait for the timeout of the backlight.
Setting it to MAX when leaving the menu is fine, but keep it on FULL as long as you are in MENU.

JuantAldea, 2023-12-15:
Yes, the failsafe is out of discussion. But it is also true that it is not returned to the configured max BL exactly when it should.

If you go above the Blit options, you will see that the BLit does not return to the configured until you are "two options away" from them, or actually, two key-presses away.

egzumer, 2023-12-15:
The button pressing functions are a mess, I know it is not perfect. I didn't want to spend more time fighting that, this was the best I could get it to be then.
I'll take a look at the solution later.

egzumer, 2023-12-15:
It is cosmetic and takes some memory space, but the operation is nicer, I like it. I tried to make it work like that before.

closed 2023-12-15 by egzumer

kittenarmy, 2023-12-15:
> I actually like the backlight at FULL when entering or traversing the menu, it's kind of a failsafe. Just imagine you are by accident setting MAX to 1 ( might happen if you think you are setting MIN [...]

If you have dark adapted eyes, using the radio with this behaviour is actually painful and not at all intuitive. It somewhat defeats the point of having option to lower the brightness. I prefer set and forget and expect the same. I suggest it be a build option like the guarding of TX on all frequencies.

egzumer, 2023-12-15:
I'm raising the stakes: 1151cf86670d5c9076f0e947ab1f73b946fabb66

Interestingly, there is more code but it takes 52B less space :)

kittenarmy, 2023-12-15:
> I'm raising the stakes: [1151cf8](https://github.com/egzumer/uv-k5-firmware-custom/commit/1151cf86670d5c9076f0e947ab1f73b946fabb66)
> 
> Interestingly, there is more code but it takes 52B less space :)

This is great. Thank you!

commit cc49a500 (2023-12-15): Fix #275 Backlight does not get back to MAX brightness after exiting backlight menu entries

Backlight is set to FULL when interacting with backlight menu entries,
but it is not set back to MAX when exiting from those entries. This
commit corrects that behaviour.