egzumer/uv-k5-firmware-custom #278: v0.20.1 Scanning, reverting to starting frequency

Handypants, 2023-12-15:
I know this is by design, but instead of both the menu button and PTT being used to end the scan and revert back to the last channel where the transmission was found, can you make it where the menu button stops the scanning on the specific frequency where the RX was found?

With the current firmware update (my first experience), it makes it difficult to use the scanning feature because when I find a clear RX, I have to quickly take note of the frequency before it starts scanning again, and have no way to stop it on the current frequency to continue listening.

Thank you for everything!

mentanah, 2023-12-15:
ScnRev - scan resume mode
CARRIER - resume scan after signal disappears
TIMEOUT - resume scan after 5 seconds pause
STOP - after receiving a signal, stop the scan

Handypants, 2023-12-15:
> I understand you and I don't. I was thinking about splitting M and PTT functionality. But you say you have to note the frequency... you don't have to. It goes to the last found frequency, so the one that you presumably had to note.
> 
> > can you make it where the menu button stops the scanning on the specific frequency where the RX was found?
> 
> It does exactly that.

Unfortunately, it isn’t working for me correctly then. This function only works when I am scanning through the channel selection, and not the frequency. When scanning through the memory channels, it works properly, I can click the menu button, and it stops on the current channel that was receiving the RX. If I am Scanning through frequencies and click the menu button, it reverts back to the starting frequency.

Handypants, 2023-12-15:
> ScnRev - scan resume mode CARRIER - resume scan 3.6s after signal disappears TIMEOUT - resume scan after listening for 5 seconds, even when signal is still present STOP - after receiving a signal, stop the scan
> 
> Set ScanRev to stop should do the job.

The only problem with this is that it takes away the ability to quickly choose which frequency to stop at after receiving a RX. For example, I wouldn’t want it to stop on a frequency that received a non audible RX because of a lower squelch.

egzumer, 2023-12-15:
> If I am Scanning through frequencies and click the menu button, it reverts back to the starting frequency.

Well, this is not how it behaves on my radio. Just tested, it reverts back to last RXed freq.

Handypants, 2023-12-15:
> > If I am Scanning through frequencies and click the menu button, it reverts back to the starting frequency.
> 
> Well, this is not how it behaves on my radio. Just tested, it reverts back to the last RXed freq.

Strange. I will re-update the firmware and see if that changes anything and I will get back to you. Thanks again.

closed 2023-12-17 by egzumer