egzumer/uv-k5-firmware-custom #282: Setting to prevent broadcast interrupting FM commercial radio

kganeshr, 2023-12-15:
This is more of an enhancement. is it possible to introduce a setting to prevent broadcasts from interrupting FM commercial radio? Sometimes people might just want to listen to the FM commercial broadcasts without interruption. Thank you.

f5gkw, 2023-12-15:
IJV firmware has this feature.
So I think it is possible to insert in egzumer.
Maybe it is related with the possibility to display only one vfo without
dual watch ( ?)
JPh

kganeshr, 2023-12-15:
[image]
I see the Dual watch is disabled when the FM is started.

ErikS-web, 2023-12-15:
As HAM we enjoy it when somebody is asking for a call. Who doesn't ?

A Tip.
Select some crazy frequncies on both VFO, start FM and enjoy..... mmhh buy a FM radio with better sound.

ps. Stereo,RDS, Super Bas... its's all not possible at this portophone.

kganeshr, 2023-12-15:
I understand the primary use of this radio is to listen to others talking. As a HAM and an avid radio listener, I  do indeed own quite some radios for FM/MW and SW listening and DXing. I thought the enhancement requested could come in handy at times when you do not feel like pulling out your big guns or not interested in carrying more than one radio. Thank you.

egzumer, 2023-12-15:
As Erik said. 
Set some frequency where there are no transmissions, which is most of the frequencies, most of the time. It doesn't have to be set on both VFOs as dualwatch doesn't work with FM radio mode.

closed 2023-12-15 by egzumer