egzumer/uv-k5-firmware-custom #283: requencies from 630 to 840 MHz

UW5EOT, 2023-12-15:
Why is it not possible to listen and transmit at frequencies from 630 to 840 MHz when fully unlocked?

ErikS-web, 2023-12-15:
Read the [manual](https://github.com/egzumer/uv-k5-firmware-custom/wiki)
and download on internet the datasheet of the BK4819.

Enjoy reading this all this weekend ;-)

UW5EOT, 2023-12-15:
> Read the [manual](https://github.com/egzumer/uv-k5-firmware-custom/wiki) and download on internet the datasheet of the BK4819.
> 
> Enjoy reading this all this weekend ;-)
> 
> P.s. this is not an firmware issue. Prefer next time to start a thread in Discussions

Should I send you a video of how this range works on other firmware?

egzumer, 2023-12-15:
> Should I send you a video of how this range works on other firmware?

Yes please.
I don't have any way of testing that and sources like [this one](https://github.com/Tunas1337/UV-K5-Modded-Firmwares/blob/main/README.md) say otherwise.

UW5EOT, 2024-02-25:
Reception at 800 MHz

https://github.com/egzumer/uv-k5-firmware-custom/assets/150297992/3ef7300e-1bb0-4f97-acb6-e10ae9fd5141