egzumer/uv-k5-firmware-custom #287: CROSS BAND audio switched A/B

AubsUK, 2023-12-16:
Firmware Version: Egzumer v0.20.1

Menu >> RxMODE >> CROSS BAND
FN+2 to switch to A
FN+3 to switch to MR
FN+2 to switch to B
FN+3 to switch to VFO

Type in a frequency just below a known (continuously broadcasting) frequency.
Press and hold */Scan, it'll stop at the known frequency and audio heard.
Exit the scan.
Press ^B to move up to the known frequency, nothing heard.

Menu >> RxMODE >> MAIN ONLY
Audo returns.

Menu >> RxMODE >> CROSS BAND
Audio gone.

FN+2 to switch to A
Audio returns.

In RxMODE >> CROSS BAND, if A is on a memory channel, B is active, if I do a search by pressing */Search, it'll stop at a found frequency and I can hear the audio.  But if I then manually tune to that same frequency (either scroll up/down, or type it in), I don't get anything even with opening the squelch (monitor).  Switching it to MAIN ONLY, it works fine.

Just tested it again, on MAIN ONLY, tune to a frequency. Hear the continuous audio, Menu >> RxMODE >> CROSS BAND and I lose it again. But, if I Press FN+2 to switch to A, I get the audio from the frequency of B, even though the arrow is pointing to the top frequency readout which has no signal.  Looks like something is broken with it.

egzumer, 2023-12-16:
Are you sure you understand what crossband is?
https://github.com/egzumer/uv-k5-firmware-custom/wiki/Menu#main-menu

AubsUK, 2023-12-16:
I clearly didn't.  My sincere apologies.

closed 2023-12-16 by AubsUK