egzumer/uv-k5-firmware-custom #289: Feature Request : Direct Mode (Point To Point)

imanuelfrederick, 2023-12-17:
Hello, can i request a "Direct Mode" feature to be added in softkey/shortcut. It really helps if the distance from repeater is far away and want to talk directly or in a repeater but want to talk just to the person directly in the area. Thank you!

egzumer, 2023-12-17:
Button `8 R`
https://github.com/egzumer/uv-k5-firmware-custom/wiki/Button-functions#front-keypad

gartnl, 2023-12-17:
Reverse mode allows you to check if you could work a person directly. If confirmed you should agree to move to a simplex frequency. But you should never work simplex on a repeater input frequency or transmit on the output with reverse turned on.

imanuelfrederick, 2023-12-17:
Okay, thanks for the response and information.

closed 2023-12-17 by egzumer