egzumer/uv-k5-firmware-custom #290: Reduce low power level

AsciiWolf, 2023-12-17:
If possible via software, please consider reducing the "L" power level to something more reasonable. Currently, it is over 2.4 W for L power. See [this video](https://www.youtube.com/watch?v=42ygQeGdTyY).

egzumer, 2023-12-17:
Use build option for power reduction

ErikS-web, 2023-12-17:
Nice Table of measurements at the end of the film

[image]

p.s. Subject POWER also mentioned in **_Reduce power_** #52

gartnl, 2023-12-17:
#egzumer Did you, or someone else, ever measure the output with the reduced option on?

gartnl, 2023-12-17:
> Nice Table of measurements at the end of the film
> 
Only one problem: look at the 27 MHz test at the end. The powermeter measures total power, including spurious and harmonics. So there is no way to be sure ( and I doubt it ) that the measured power is transmitted on the correct frequency.
Only way to really see what is going on is a spectrum analyzer.

egzumer, 2023-12-17:
I've got only [this](https://github.com/egzumer/uv-k5-firmware-custom/issues/66#issuecomment-1786868844):
> measurement data before firmware: H =4.5 , M=3 w ,L=1.2 data after firmware : H=4.5 w , M <1 w, L < 0.5w . The measured power in low mode is about 200 MWt, the switch device shows only a slight presence of transmission. Thank you so much for your help!

That is what he provided and I didn't want to push him, but it would be nice if someone could do some tests.
I don't know what are the differences between different units. How good the factory calibrations is. What are the different values across spectrum, etc.
I don't have power meter and don't have experience with this topic that is why I didn't want to touch it.

egzumer, 2023-12-17:
https://github.com/ludwich66/Quansheng_UV-K5_Wiki/wiki/Measurements

egzumer, 2023-12-17:
The Qunsheng calibration itself contains 3 calibration points for beginning, middle and end for each of the bands.
I will also add calibration editing to chirp driver.

gartnl, 2023-12-17:
> https://github.com/ludwich66/Quansheng_UV-K5_Wiki/wiki/Measurements

That is real good info! And, again, it shows that the HT should not be used outside of the design.
I'm tempted to get myself an SA for Christmas just to see ...

Hunterfnet, 2023-12-17:
The stock V.26 version is much old for compare and have bad performance , better try with a new online customs firmwares , the official that receives with the walkie or new stock versions for compare with the egzumer version and flash with k5prog and do not use the old UVMOD ver V.26 for reflash . This comparison says only that the old V26 firmware is obsolete out of date

Hunterfnet, 2023-12-17:
For AsciiWolf
The only way for reduce the power with egzumer firmware for now is transmit with N Narrow mode and L together almost on real distances (tested) and working . On others firmwares you can do others things and/or chose the ultra Narrow option N- that here is not the case.

ErikS-web, 2023-12-17:
He uses version 1.26 and not 1.27, on which all custom firmware is based. _But I'm not sure if that is an issue._

Looks like testing is happening at 145.475 and 433.575
I measured at 145.0000 with upper VFO. Center of the 2 meter band.

[image]

With a powermeter that works up to 150 Mhz (CN620A). Cannot measure the 70 cm.

mentanah, 2023-12-20:
... and another fresh measurement from my dad recently with QS = egzumer v.19 UV 5R Plus from Quansheng with short and long (38cm) antenna and dummy load:
[image]

maxibor, 2024-02-14:
An implementation of an even lower power setting can be found in this fork https://github.com/armel/uv-k5-firmware-custom

Here: https://github.com/armel/uv-k5-firmware-custom/blob/3e6c2959aeb5fc1b6c250766c8c5f95e601b7ad8/radio.c#L475

Apparently promising a low power setting as low 20mW.

```c
#ifdef ENABLE_FEAT_F4HWN
	// make low and mid even lower
	if (pInfo->OUTPUT_POWER == OUTPUT_POWER_LOW) {
		for(uint8_t p = 0; p < 3; p++ )
		{
			switch (gSetting_set_low) {
				case 0:
					Txp[p] = (Txp[p] * 4) / 19;
					break;
				case 1:
					Txp[p] = (Txp[p] * 4) / 13;
					break;
				case 2:
					Txp[p] = (Txp[p] * 4) / 10;
					break;
				case 3:
					Txp[p] = (Txp[p] * 4) / 7;
					break;
				case 4:
					Txp[p] = (Txp[p] * 4) / 25;
					break;
			}

		}
	}
```

Maybe the original author of this feature @armel can tell more about the real output power in the end ?
I guess it's especially interesting to make it more PMR446 compliant (TXP < 0.5W)

nosektomas2004, 2024-08-27:
Hello
any chance this will ever be available? Set "custom" performances for LOW and MID?
Personally, I would find it best if 
LOW = 0.5W 
MID = 1W
HIGH = about 5W (maximum posible)...

AsciiWolf, 2024-08-27:
Same here. Although in my case, the "sweet spot" would be 0.5W for LOW, 2W for MID and MAX (5W) for HIGH. :-)

BruMcD, 2024-09-15:
Just read this page

Possibly helpful things to know

- The output power 'number' for the lower centre and upper end of each band can be seen  in CHIRP calibration TX power
- A 'calibration number for Low Centre and Upper on F3 and another for F6 is inserted at the factory for High Power levels only
- The number varies as really you are setting a PA stage gain figure to compensate for a particular hardware.
- The number can be set over a wide range but in practice values for high power tend to fall between 110 and 135 on Egzumer Chirp
- On F3 the Medium figure is always 105 and the Low is 95.  The 'scale' is different to High power though
- On F6 the Medium figure is always 100 and the Low is 90.  The 'scale' is different to High power though
It is perfectly possible that output will stop at a very low setting and nothing useful be achieved by increasing the number above a certain point.

Internally the radios uses the reference number values to create rough PA gain curve to apply to any requested transmit frequency in the band for correct power.  The radio has no power or SWR sensing so it cannot be asked directly for 3.5Watts

If you have a decent power meter and considerable patience you can change the factory high power settings a bit or reduce the Med or Low power.
Unfortunately the 'curve' means that none of the power setting numbers work entirely independently.  Upper Freq is most independent but changes to Centre and Low see the whole set of associated power outputs see-sawing (e.g reduce value power goes up) in some strange cases so it is necessary to try values and retest all three frequency spots and at any most likely operating frequencies.     Before anyone asks **NO I cannot tell you your settings or Your best revised settings or the initial settings you now wish you had recorded **
e.g. On the last one I adjusted reducing F3 band Low power from 3.5 to  2.5Watts meant changing 95 95 95 to  70, 68 60 but mostly the right answer on low power is between 100 and 65 but not very predictably

**Now the elephant in the room**.   
Measuring RF power at VHF /UHF is quite hard.  Few consumer grade power meters will be right.  Mostly they say the accuracy is 10% of FSD value meaning on a 20W scale +/-2W   At half scale and upper /lower frequency range limits all bets are off but be prepared for +/- 3W.     Clearly Hams and hobbyists don't appreciate the - side so  it has long been traditional that Ham grade power meters over read Watts by 10 - 20% in order to 'improve user satisfaction' and more important reduce return rates

On YouDude you see many cheap digital power meters being treated like prophets.   I have 2 as they are convenient but when benchmarked against a panel of Expensive Daiwa VHF/UHF meter,  Spectrum analyser via a 10W 50.9dB (50dB nominal verified on 2 different VNA's)   attenuator etc the fancy digital ones always read highest    Typically 10 to 20% high just like the old days

Measuring a factory fresh radio as accurately as possible and comparing 3 radios I see the factory goal is High power of around 4.5W into a dummy load across the F3 and F6 (VHF /UHF) bands mentioned on the sets identity sticker.   A pass is probably between 4.0 and 4.9 and output can droop at band edges due to transistors and LPF's 
As mentioned all the Factory Medium and Low numbers I've seen are identical from set to set.  They seem standard and not individually calibrated on any set I saw.
Likewise standard values are applied for other bands not 'on the sticker'
Looking at the output spectrum - Reducing power I remember I saw the main carrier reduce just as expected but the harmonics less so.  At 4.5w out I saw -46dBc (relative to carrier)  which is a fairly good result and I think meets the FCC rules but on lower power c 2.5W let's say its getting closer to those limits at least on the one I tested.  Both levels were within the Spectrum analyser's sweet spot but of course a more expensive model would provide more confidence.

As the factory is clearly not going to calibrate med/low powers let alone check every low power frequency with a spectrum analyser the answer would appear to be to pitch the standard Med/Low settings at quite a large % of high power.  
As we have all seen this is what they have done and I'm not seeing another explanation really 

Finally on High power.  It is possible to push the power above 5W.   The radio is most likely internally dissipating more Watts than it's radiating at that point and frequencies drift, components overheat etc.   The range of 5.5 W is almost identical to 4.5W and the difference will never be heard in practice.   
Making sure you have 4.5W is useful    Cooking it for 6w is definitely a bad idea if you actually plan to talk on it  😇