egzumer/uv-k5-firmware-custom #292: splatter noise at AM reception

mentanah, 2023-12-18:
Sometimes there is a splatter noise at AM when receiving a signal between S2 and S3. In my opinion this is due to fast toggeling the gain of the LNA or PGA in the background.
Is it possible to ad a hysteresis to prevent bouncing gains when signalstrength is on edge of a gain step?

Maybe im totaly wrong with this explanation, please delete this issue then.

egzumer, 2023-12-18:
There is a hysteresis. Build ithe FW with ENABLE_AM_FIX_SHOW_DATA or ENABLE_AGC_SHOW_DATA. It will show you the gain settings.
You can also see what are the gains in the finetunning page of spectrum

mentanah, 2023-12-18:
When I'm entering the finetunning page of spectrum the splatter either stops or gets really worse with strong oszilating bar Graph.

egzumer, 2023-12-18:
You have to lower the squeltch level to the minum there. You can still use the other options I mentioned.

kamilsss655, 2023-12-22:
@mentanah I've noticed it that as well when listening to SW broadcast stations.
Can you test with https://github.com/kamilsss655/uv-k5-firmware-custom/releases/tag/v.05 ?

It uses old @OneOfEleven gain tables.

mentanah, 2024-02-05:
I analyzed it a bit. watching the LNA, LNAs and PGA gains you can see the following:
When the gains reached 2, 7, 7 the next step would be to increase the LNA to 3 and to decrease the other. This happens very fast and when switching to LNA 3 the sound gets distorted for 100 ms until it switches back to LNA 2. This goes forth and back so quickly you can hardly see on display.

Since the LNA as a low noise amplifier produces better signal quality than the PGA programmable gain amp, the LNAs should used up before increasing the PGA not vice versa.

mentanah, 2024-02-05:
> @mentanah I've noticed it that as well when listening to SW broadcast stations. Can you test with https://github.com/kamilsss655/uv-k5-firmware-custom/releases/tag/v.05 ?
> 
> It uses old @OneOfEleven gain tables.

I want to stay at egzumer as my favorite software, maybe he can help us with this?
Unfortunately sometimes this happens in FM as well.

egzumer, 2024-02-05:
At some point I noticed that switching LNAs gains in the spectrum gives different signal strength differences than 1o11 assumed, so I changed the values and regenerated the whole gains table. This gave issues as was mentioned in #212. 
Coilin (g4eml) did some tests and modified the table and this is how it is now.

The initial change might have been unnecessary and maybe moving back to 1o11 table is actually a good idea, I don't know. I was planning to make simple low power AM transmitter for testing but I didn't do that yet.

> Since the LNA as a low noise amplifier produces better signal quality than the PGA programmable gain amp, the LNAs should used up before increasing the PGA not vice versa.

It is. Notice that the gains are minus values, so it is more of an attenuation table, and LNAs gain is decreased at the last possible moment.
In other words for low signals LNAs is always set to the highest gain possible.
The issue as you said is when LNAs switches from 3 to 2 or from table index 34 to 35. It looks like the actual overall gains are different and 34 has actually a higher gain than 35 and it should be the other way round and this causes oscillation between them.

https://github.com/egzumer/uv-k5-firmware-custom/blob/1098e8c38655716fbe838ad968650603dbd76202/am_fix.c#L104-L145

egzumer, 2024-02-05:
You can try the old gains restored: https://github.com/egzumer/uv-k5-firmware-custom/actions/runs/7790891496 (see the artifacts)

mentanah, 2024-02-05:
Thank you egzumer for this good explanation and your efforts supporting us so quickly.
Actually I tried your restored old 1o11 gain table - and yes it is perfect for me now! No sound glitches anymore. 
I remember some people complained about signal quality on very strong airplane reception in v0.19. but for me it's perfect now! Can I buy you a coffee ☕?

egzumer, 2024-02-05:
Strong signals problem had different source described here: https://github.com/egzumer/uv-k5-firmware-custom/wiki/Development-notes#1o11-am-fix-issue-with-strong-signals

As to the coffee, thank you, but I don't expect any reward, the software is a product of work of many people, we all did it for free. I thought about turning on donation button, maybe I will. Until then simply enjoy the software and don't feel obligation to show any appreciation.