egzumer/uv-k5-firmware-custom #302: Manual for custom firmware - wrong info

FrostyMisa, 2023-12-20:
I want to mention, looks like you make a change for the Spectrum Sweep screen behavior, because now the selected frequency is in the middle of the spectrum, not at the beginning like in version before, and this is not mention in the manual.

I like this behavior more, because you scan "around" the frequency you set, not "from" the frequency and after. 

And I want to thank you very much for your custom firmware, all the settings and functionality you change and add to the Quansheng UV-K5 is great and exactly what I imagined and I want for it!

mentanah, 2023-12-20:
> And I want to thank you very much for your custom firmware, all the settings and functionality you change and add to the Quansheng UV-K5 is great and exactly what I imagined and I want for it!

Yes, definitely!

ErikS-web, 2023-12-20:
Thanks. Wiki updated

With firmware v0.21: The current VFO/Memory frequency is the centre frequency of the spectrum sweep

closed 2023-12-20 by egzumer