egzumer/uv-k5-firmware-custom #304: Scanning error - Gets stuck in a loop between two channels

MikeGriffinReborn, 2023-12-20:
Scanning will get stuck into a end loop between two channels.

Steps to reproduce:

1. Start Scanning (I have the long push on "Button 1" set to start scanning)
2. Hit button 1 again - it goes into RX mode on whatever channel it's on during the scan
3. The repeat Step 1 again, long push on "Button 1" and you will see it bounce back/forth between two channels forever

ErikS-web, 2023-12-20:
Hi Mike... 

can you be more specific. I can't reproduce.

Function(s) of [Button 1 ](https://github.com/egzumer/uv-k5-firmware-custom/wiki/Button-functions#1-band)

JuantAldea, 2023-12-20:
Hi, @MikeGriffinReborn, would it be possible you that have added two channels to the "VFO scan list" so you are bouncing between the two?

MikeGriffinReborn, 2023-12-20:
This bug could have honestly been already present in the shipped firmware, but here goes ...

[image]

Notice that I have side key 1 button set to scan, now watch this video ...  

NOTE: You don't have to register, just choose 1MB per second download ... it works, I just tried it ....

[Download Here](https://ufile.io/19w7zu17)

MikeGriffinReborn, 2023-12-21:
Ok, thanks, I gotcha, it's a strange scenario anyway, thank you for looking into it !

closed 2023-12-21 by MikeGriffinReborn