egzumer/uv-k5-firmware-custom #310: backlight dont turn off on spectrum screen

Hunterfnet, 2023-12-21:
as said on title , another fantastic issue

gartnl, 2023-12-21:
This is by design. You can turn it of with sidebutton i assigned to backlight.
I don't regard it an issue as I want to see where signals are and impact on battery is neglictable.

Hunterfnet, 2023-12-21:
i assumed too your response , then the design is bad when i set a backlight default time it's a generic design too , dont need that or make mistakes as generic designs. That's the easiest way

gartnl, 2023-12-21:
Well, compile your own firmware than .... The design is not bad. The whole idea behind a bandscope or spectrum analyser is to look at it so you would expect the backlight to be on.  And you still have the option to turn it off. 

Perhaps it is your understanding of English but a remark like 'another fantastic issue' does not sound nice to egzumer. It sounds like you blame him. But don't forget he is putting a lot of time and effort into this and all for free!

Hunterfnet, 2023-12-21:
i'm here reporting issues , as you can see , not for try sell a mistake like a fantastic evolution and not for compiling nothing . If i compiles something i try make the best before i place like a big software version. And certainly is another  issue as you can see a lot here, nothing more.
On other face , all engineers know about what i'm speaking , you seems that not. Maybe i'm a bad day too today , like all people on any day ;) . Nice day and good work

mentanah, 2023-12-21:
Thank you egzumer, for your excellent work and frequently releasing new updates. 
You are doing a great job for a lot of people here for sure!

Hunterfnet, 2023-12-21:
on my country this is called "palmeros" 
I don't placed wood on fire guys , for same rule dont need a timeout for backlight . When one is on stand by on a frequency , same on the spectrum , when the signal is hard enough the light turns on and heard something. Pls. don't try make the people little for make feel yours better , that dont works . Cya

egzumer, 2023-12-21:
This is how it was designed by fagci, I don't see a reason for changing that

closed 2023-12-21 by egzumer