egzumer/uv-k5-firmware-custom #315: XBAND offset direction

gripitoy, 2023-12-21:
Setting a crossband memory in egzumer version chirp does reflect in the radio when uploaded.

ISS for instance has an uplink of 145.990 and a downlink of 437.880. What we do is set a -291.89. When uploaded to radio, the offset is retained but the direction is not.

Only by manually changing the direction in the radio and overwriting the memory can it be saved.

Thank you for the great firmware.

P.S tried uploading the video but it says file is too big. I'll share a link for it here.

https://drive.google.com/file/d/1Ic_bNohhGkCq1bi6d_XNcLmGfaoYu31O/view?usp=drivesdk

gartnl, 2023-12-22:
Tried this and it works as expected.  Do you set duplex to minus and offset to 291.890 or just offset to -291.890 .  That last option does not work.
Also be sure to use the latest driver for chirp.

gripitoy, 2023-12-22:
I'm using the former. Offset set to 291.890 and set duplex to -. No
worries, thanks for getting back to me and letting me know this is how it
should be.

 Griffith Boniel
Customer Service Professional

egzumer, 2023-12-22:
fixed https://github.com/egzumer/uvk5-chirp-driver/commit/ceb028ded0d5288131c40500cd37a567a3f54a9a

closed 2023-12-22 by egzumer