egzumer/uv-k5-firmware-custom #318: Upconverter mode sugestion

CALIGV, 2023-12-22:
It would be great to implement the upconverter mode (Rx) in order to program a frequency manually, there are other firmwares that have it but in a fixed mode (50-125MHz) but it is not very useful for those of us who have homemade upconverters or with another non-standard frequency .
It would be super useful

kamilsss655, 2023-12-22:
I had the same issue as my upconverter runs at 40.003Mhz. :)

In the end I've implemented it myself.

You might want to check out: https://github.com/kamilsss655/uv-k5-firmware-custom

CALIGV, 2023-12-22:
thank you!
I just tried to flash the latest firmware that you suggested but without success, the process is aborted at 70%, both using the Windows software and the web browser

kamilsss655, 2023-12-22:
Interesting. Did you try to upload `firmware.bin` or `firmware.packed.bin` from https://github.com/kamilsss655/uv-k5-firmware-custom/releases/tag/v.05 ?

CALIGV, 2023-12-22:
Yes, correct, it was firmware.packed.bin and it only reaches 70%, the radio was soft bricked but I flashed the egzumer0.21 and it revived normally, then I tried firmware.bin but the latter gives a version error and cannot be flashed

kamilsss655, 2023-12-22:
Interesting, I've uploaded it with k5prog (https://github.com/OneOfEleven/k5prog-win).

CALIGV, 2023-12-23:
I was finally able to flash the radio. connector, had dirt and had intermittent contact.
I am testing the Rxoffset function and it is great, the only detail is that the step is 1kHz, any possibility of increasing the step to 100 or 10Hz?
my upconverter is 39,994,540 lol

kamilsss655, 2023-12-23:
O wow that is bad luck with that LO frequency :)
Sure I can add this in the future, just make a feature request issue in my repo so I don't forget.

Btw how was your experience with this fw? Any issues?

CALIGV, 2023-12-23:
My upconverter is homemade, so for the LO I used a small 40MHz crystal (colpitts jfet oscillator) but naturally it is not exactly at the frequency and depending on the ambient temperature it moves around 100+-Hz but with very good stability, and the adjustment in steps of 100-10Hz helps a lot in tuning stations in SSB
Regarding the firmware, it works well but it has some details, I have detected small freezes using the UP-Down buttons to advance the frequency, in AM the stations can be heard perfectly, sometimes the audio in AM mode begins to degrade over time, sudden changes in volume or choppy audio, all in Rxoffset mode

kamilsss655, 2024-01-01:
@CALIGV You can fine tune this setting in even smaller steps by changing `Step` setting to `0.01kHz` and then use `UP/DOWN` buttons to alter `RxOffs` in increments of `0.01kHz`.

@egzumer apologies for off-topic.

ErikS-web, 2024-01-02:
If you realise it. Pse consider... "Upconverter Active should block TX"

I thinks that a good idea to protect the upconverter from damage.