egzumer/uv-k5-firmware-custom #325: minor change of description in makefile

gartnl, 2023-12-24:
In the makefile the option
"ENABLE_COPY_CHAN_TO_VFO       := 1       copy current channel into the other VFO."
gives the impression that the channel is copied to the other VFO. But it is copied to the same VFO.
E.g. if i use a channel in VFO A it is copied to VFO A, not VFO B.

egzumer, 2023-12-25:
VFO means frequency mode. There is also channel mode, hence chan to VFO.
I didn't choose that naming. It is hard to even name it properly, there is upper channel, lower channel, frequency channel, "channel channel".
It is what it is.

gartnl, 2023-12-26:
I understand your point, naming these features has become diffuse. Maybe I can clarify some things, but that is from a strict HAM view. But pls understand: in no way this is meant to be leading for your project.

From a HAM/technician perspective there are 2 VFO's, as in the old analoque era: VFO-A and VFO-B.
A VFO can be regarded as a seperate receiver, it is the Variable Frequency Oscillator for a mixer.
Wether you set the VFO freely by hand or from a fixed value in memory, it is still the same VFO.

Channels only excist in bandplans and define fixed points in frequency assignement to be used. The step size ( increment per up/down ) is often set equal to the channel distance. Like in CB where the stepsize of 10 enables one up/down press to go to the next channel. Channels can be numbered, CB channel 14 is the 14th channel/step counting from the beginning of the CB band.
Often the pre-programmed frequencies in memory are not displayed as frequency but as the channel number. So, to keep it in C, the channel number is the index for an array of stored frequencies: CB[14] = 27.125 .

On the HT the upper display is regarded as VFO-A and the lower as VFO-B. So for a HAM or trained radio-operator when tuning VFO-A 'the other VFO' is VFO-B.

egzumer, 2023-12-30:
I read again your suggestion and yes the description was misleading, I concentrated on the option name itself at first. I changed the description. Thanks.

closed 2023-12-30 by egzumer