egzumer/uv-k5-firmware-custom #326: Power Level Issues (changed High to Medium)

JohnRobson, 2023-12-24:
After applying the Firmware_IJV_V2.3R5.bin, it seems that this firmware is reading "High Power" Settings as Medium Power.

All my channels changed from "H" to "M", the CHIRP or "PSCPS.exe" shows "High", but the radio shows "Medium".

If I change Manually (in the radio) to High, it now shows "H", but CHIRP reads as "Low".

ErikS-web, 2023-12-24:
Here we discuss EGZUMER firmware.

Do you have the eeprom-backups (made with K5PROG). You might to restore that.

egzumer, 2023-12-25:
You messed something up in your testing/description. Power is saved on 2 bits. This is not changed from the beginning. It is still the same as quansheng implemented it. It is not possible for the CPS or CHIRP to show different values than what is saved in the channel on the radio.
I don't know how IJV saves his power settings, if he changed it you should talk to him.

closed 2023-12-27 by egzumer

JohnRobson, 2023-12-28:
> Here we discuss EGZUMER firmware. Which is open source. No idea what other firmware does.

I got it from here:

https://egzumer.github.io/uvtools/?firmwareURL=https%3A%2F%2Fwww.universirius.com%2FSirioArchive%2FMateriel_pr_site%2FFirmware-IJV%2Ffirmware_IJV_V2.3R5.bin

egzumer, 2023-12-28:
> I got it from here:

Which is not a part of this repository, it is not even a part of my other repo. Read carefully what that links adress consists of. You should talk to the person that gave you that link. I have nothing to do with that.

JohnRobson, 2023-12-29:
@egzumer ok, maybe I understand.  YOUR repo "egzumer.github.io" allows people to "publish" firmwares? (like this one "universirius.com"?)

It would be much faster if you just explained that, since the MAIN URL "egzumer.github.io" is YOURS.

egzumer, 2023-12-29:
> YOUR repo "egzumer.github.io" allows people to "publish" firmwares? (like this one "universirius.com"?)

No it is a firmware uploading tool that you can pass a link to a firmware and it will load that firmware. If someone sends you a link to a shared file on his google disk you don't call google, because it starts with "google.com", do you?