egzumer/uv-k5-firmware-custom #344: SSB

ClintHennessey, 2023-12-28:
WOULD BE NICE TO ADD SSB TO THE DEMODU BAND SELECTION AND W/N WITH WIDE + WIDE - AND NARROW+ NARROW - FOR SSB AND USB RX

ErikS-web, 2023-12-28:
He Clint CAPS-lock defect ?

There is NO SSB; see Menu 13. 1 `Demodu` - demodulator mode, default is FM, AM/**USB can be used for listening only**

Can you be more specific with what you mean ?

ClintHennessey, 2023-12-28:
Good evening
sorry about that was just wondering if could be done as seen on one of the other firmwares that person had SSB and had the choise of N N- W W-
here is the link to the bin file and the youtube
on what im trying to explain
(url)https://www.universirius.com/SirioArchive/Materiel_pr_site/Firmware-IJV/firmware_IJV_V2.3R5.bin
(url)https://www.youtube.com/watch?v=2U9zHlnLmmA)
Kind regard