egzumer/uv-k5-firmware-custom #356: PowerOn Password and SOS flashlight

Kemon22, 2023-12-30:
Hey, maybe I'm doing something wrong and/or I'm misunderstanding, but I have two problems. 

> 1. The flashlight in the third mode (third click) should transmit an SOS signal, I think, but unfortunately it transmits 4x short -> 3x long -> 4x short.  
> 2. Unfortunately, for some reason the poweron password simply disappears from the configuration, even though I set it in the software.  After uploading it to the device, I don't have this password, and after reading the data again, the password is no longer there.

egzumer, 2023-12-30:
Show a video of the SOS

egzumer, 2023-12-30:
What version does it show on the radio startup?

crazy4volvo, 2023-12-30:
IMHO the only issue with SOS is lack of pauses between letters - instead of "tititi tatata titi"  led transmits "titititatatatititi".

egzumer, 2023-12-30:
There are no pauses, read up on how SOS should be transmitted

Kemon22, 2023-12-30:
Sure, I'm sending it now.  However, when I have recorded it, I noticed that the count of blinks is good, but there is no delay between the "letters".  Overall, this flashlight is not a big deal.

Kemon22, 2023-12-30:
Uploading VID_20231230_154543.mp4…

Kemon22, 2023-12-30:
But still, I still have some problem with PowerOn Password.
Ofc it can be my fault but idk why it doesn't want work.

egzumer, 2023-12-30:
You didn't send anything. At least tell what firmware you are using. It seems that you are using some old firmware and a one that has no power on password functionality built in.

crazy4volvo, 2023-12-30:
> read up how SOS should be transmitted

Oops, my bad. One must live long to learn...

Kemon22, 2023-12-30:
Well, from about 15 minutes I have "Uploading VID_20231230_154543.mp4…" so maybe it will send in a few years xD
My firmware is EGZUMER v0.21 running on UV-K5(8)

egzumer, 2023-12-30:
There is no power on password built in the official release. Did you build your own version?

Kemon22, 2023-12-30:
By "official" you mean your official or the manufacturer official?
And no, I didn't build anything own.

egzumer, 2023-12-30:
You've got your answer then. There also should be no password field in chirp if you download configuration from the radio.

Kemon22, 2023-12-30:
Okay then. Maybe you want to try implementing it? :D The original software allows to enter a password. Obviously, it's not a great security measure, but it limits access to the device a little bit.

Kemon22, 2023-12-30:
And thanks for answers. <3

egzumer, 2023-12-30:
It is implemented. It is not included. If you want that feature you have to do the build on your own. All you need is in the readme.

closed 2023-12-30 by egzumer