egzumer/uv-k5-firmware-custom #359: Enhancement : spectrum analyzer assignable to Side button II

MartinR577, 2023-12-31:
I like to keep the NOAA functionality  with a custom firmware build so F+5 activates NOAA but then I  can't activate the spectrum analyzer. please consider making it one of the options for the Side II button long long press

CharlesTerrell, 2024-01-05:
I would be interested in this. Mine has a build with NOAA re-enabled, because it's actually useful here. But then where is the spectrum analyzer..?

User interaction for the weather alert scan feature seems arcane and unreliable. Makes me wonder if there's a better way to do it, given the limits of the device.

ErikS-web, 2024-01-06:
NOAA is only of interest in a 'smart part' of the world !

Remoran, 2024-02-02:
This would be a desirable change for me as well. If implemented, I'd be willing to help test proper functionality of both features inside the US.

bogdanbrudiu, 2024-02-15:
hey, pls check [https://github.com/bogdanbrudiu/uv-k5-firmware-custom/releases](https://github.com/bogdanbrudiu/uv-k5-firmware-custom/releases)
If everything is ok I will create a PR

bogdanbrudiu, 2024-02-16:
Yeap, my bad, I enabled F5 (NOAA) + SPECTRUM and created a new release pls have a look

MartinR577, 2024-02-16:
> That looks good now. Having NOAA channels back on F5 makes more sense
> instead of two spectrum buttons.
>
> I do see the NS icon in the corner (make sure to enable it in settings
> [option 23 here] for anyone not seeing it), but haven't had any warnings
> yet in my area to truly give it a go.

It looks like it's works great, Thank you

ErikS-web, 2024-02-18:
Compliments for this TEAM-work 👍

CharlesTerrell, 2024-02-28:
Thanks so much for doing this!

Just to be sure I understand how to use this correctly: Enabled weather scan while on B, saw NS appear, changed to A, started channel list 1 scan. Now it's scanning my channels while NS still shows in the corner. Is it actually doing both scans at the same time? Channel scan seems to be working as usual, but we have no convenient severe weather at the moment to test NOAA alerts.

Remoran, 2024-02-28:
I am not totally sure how the NOAA scan feature works, so I'm just going to vomit some information here. My unit is currently using bogdanbrudiu's "pre2" release firmware.

The following is taken from the user manual:
> ● Press F+5 to enter or exit NOAA Weather Alert.
> ● This radio could receive 10 NOAA channels.
> ● This mode could be set through Menu 49 NOAA_S.

I notice the following behavior..

NOAA scan **off** (via menu setting 23): F5 will take me to the pre-programmed NOAA channel selection of N1-N10. Navigating to my location's active channel will turn the LED solid green, display the signal bar, display RX, but not actually play any sound. Turning on the monitor function will toggle on constant audio. Not sure why that's necessary. It should act the same as normal memory channels imo.

NOAA scan **on** (via menu setting 23): F5 will take me to the pre-programmed NOAA channel selection of N1-N10. "NS" is now displayed in the corner and the LED starts slowly flashing green in a fairly inconsistent pattern. No sound is played while on any channel, and all of the NOAA channels (active or not) yield the same results. Turning on the monitor function will toggle on constant audio. If I had to guess, the radio is now scanning through every one of the NOAA channels listening for some kind of alert tone and the flashing green light is when it passes my local one with a signal but no tone was heard.

I am currently under a wind advisory that the station is constantly talking about, yet the scan hasn't alerted me to this by playing sound. A few days ago we had a thunderstorm warning and I came back to the radio after ~5 minutes with the audio playing. Maybe it only activates for certain types of broadcasts? It also never turned the audio off after it began. I will test again during more severe weather.

CharlesTerrell, 2024-02-29:
Weather radio transmits 1050 Hz tone to get attention. A radio with "alert" capability causes an alarm to sound when it detects the incoming tone. Like you said, I assume that our radios cycle rapidly through weather radio freqs, listening for that 1050 Hz attention tone. That's why it caught my attention that it was showing NS while also doing a normal channel scan. That's a lot of freq change activity to coordinate all at the same time. Please correct me if I'm wrong about that though.

Along with the tone they also transmit a short data burst encoded with more specific information about location and hazard type. It would be amazing to decode and use this info, but I'm still learning how it works. I think it involves phase-shift modulation and/or complex audio decoding. Possibly beyond UV-K5 capability without an external decoder.

USA and Canada transmit periodic alert tests. CA tests the data burst every Wed around noon, and the alert tone on first Wed of month. US tests at 11am every Wed. (Sources: NOAA and Weatheradio Canada web sites.) Bermuda and Mexico use the same basic system (MX focus is seismic events), but I have not found clear info about their testing schedule.

I set a calendar item for myself next week to test with my radio. I want to know if weather alert really works at the same time as memory channel scan. My guess is that it would fail if already listening to an active signal, but that's only a guess.

There must be a faster and more reliable way to test this. Maybe fake a very low power alert signal over an attenuator or dummy load? Wouldn't want to set off weather radio alarms around my neighborhood though. That could cause unwanted attention.

salokinrekced, 2024-05-02:
Any word on if this can be added to the main release? It works fantastic, the only tiny minor gripe is that the longpress 5/NOAA key doesn't activate the function like other longpresses do, and also that when exiting the NOAA function by pressing F+5 again puts you into Frequency mode rather than the last used mode.