egzumer/uv-k5-firmware-custom #366: Channel scanning problem

mfarikan, 2024-01-01:
There is a problem in channel scanning. It scans only one scan list, not two scan lists. Menu 20 & 21 > Slist1 and Slist2 look the same.

egzumer, 2024-01-01:
It seems to me that you didn't read the manual.

mfarikan, 2024-01-01:
I read it, but I think you haven't checked the problem

mfarikan, 2024-01-01:
[image]
[image]
[image]
[image]

egzumer, 2024-01-01:
Ok I see, sorry.

mfarikan, 2024-01-01:
It's okay, the important thing is that you're interested. 73.

commit ed5ae784 (2024-01-01): Fix #366: Scan list 2 ignored everywhere
closed 2024-01-01 by egzumer