egzumer/uv-k5-firmware-custom #373: ScnRng not working when compiled with ENABLE_VOX = 0

majonezz, 2024-01-03:
Hello, pretty nice piece of software going on here, congrats :-)
As in the title, i just find out that the Range Scan function is not working when VOX is disabled at compile time. 
ScnRng not appearing at all when "5" is longpressed at the main screen. Both VFOs in frequency mode. 
Regards, Michał.

JuantAldea, 2024-01-03:
Looking at the code, it seems 7 is the key you are looking for when VOX is disabled.

https://github.com/egzumer/uv-k5-firmware-custom/blob/ed5ae7842bae5c77008fe5b441d44fdac2fb987d/app/main.c#L237-L244

majonezz, 2024-01-03:
Well, thanks for that. It works with "7" as expected. I'm not that familiar with the code yet. I took the wiki article too serious, where it stands that "5" is for a range scan. Thank you.

closed 2024-01-03 by majonezz

JuantAldea, 2024-01-03:
> Looking at the code, it seems 7 is the key you are looking for when VOX is disabled.
> 
> https://github.com/egzumer/uv-k5-firmware-custom/blob/ed5ae7842bae5c77008fe5b441d44fdac2fb987d/app/main.c#L237-L244
> 
> Such change in controls seems unnecessary though, at least at first glance, given that Key_5 "else" part becomes empty when VOX is disabled.
> 
> https://github.com/egzumer/uv-k5-firmware-custom/blob/ed5ae7842bae5c77008fe5b441d44fdac2fb987d/app/main.c#L204-L232

cc: @egzumer

egzumer, 2024-01-03:
Yeah, this comes from 1o11, she was implementing requests from users, and different people wanted different buttons. This needs some care.

The wiki manual is for official release only.

JuantAldea, 2024-01-03:
> Well, thanks for that. It works with "7" as expected. I'm not that familiar with the code yet. I took the wiki article too serious, where it stands that "5" is for a range scan. Thank you.

Is it? The 7 was completely unexpected for me :D