egzumer/uv-k5-firmware-custom #374: buring with stlink V2

losehu, 2024-01-04:
HELLO, EGZUMER
I want to use stlink V2 for burning
I modified the first two lines of gd32g030.cfg:
Transport select hla_ SWD
Adapter speed 100
Then use openocd to execute
Sudo openocd - c "bindto 0.0.0.0" - f interface/stlink.cfg - f dp32g030.cfg - c "writd_image firmware.bin 0; shutdown;"
But prompt:
Open On Chip Debugger 0.11.0-rc2
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Force hard breakpoints
Info: The selected transport tool over low level target control The results may differ compared to plain JTAG/SWD
Info: clock speed 100 kHz
Info: STLINK V2J38S7 (API v2) VID: PID 0483:3748
Info: Target voltage: 3.273879
Warn: UNEXPECTED idcode: 0x0bb11477
Then OpenOCD terminated
I set CPUTAPID
Still useless, do you have any relevant suggestions?
  I am sure my connection is correct and STLINK is working properly 
[image]

losehu, 2024-01-04:
I got the same results on both Windows and Debian

closed 2024-01-04 by losehu

egzumer, 2024-01-04:
Did you get it to work? I also have stlink, but I have never felt like messing with it. Could you send a photo with connected wires?

losehu, 2024-01-04:
> Did you get it to work? I also have stlink, but I have never felt like messing with it. Could you send a photo with connected wires?

[image]
I have solved the problem by modifying:
transport select hla_swd
swd newdap $_CHIP_NAME cpu -enable -expected-id $CPUTAPID
AND ADD:
set CPUTAPID 0x0bb11477

Lar-Sen, 2024-01-05:
Here you have a ready to use config file for OpenOCD using STLinkv2 probe:
https://github.com/Lar-Sen/Quansheng_UV-K5_Kitchen/blob/main/unbrick-toolkit/OpenOCD/target/dp32g030.cfg

losehu, 2024-01-05:
> Here you have a ready to use config file for OpenOCD using STLinkv2 probe: https://github.com/Lar-Sen/Quansheng_UV-K5_Kitchen/blob/main/unbrick-toolkit/OpenOCD/target/dp32g030.cfg

******** Launching 1 shot unbricking process with OpenOCD (STLink v2)...
.
Open On-Chip Debugger 0.12.0 (2023-10-02) [https://github.com/sysprogs/openocd]
Licensed under GNU GPL v2
libusb1 09e75e98b4d9ea7909e8837b7a3f00dda4589dc3
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 960 kHz
Info : STLINK V2J38S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.241744
Info : [DP32G0xx.cpu] Cortex-M0 r0p0 processor detected
Info : [DP32G0xx.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for DP32G0xx.cpu on 3333
Info : Listening on port 3333 for gdb connections
[DP32G0xx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc0000000 pc: 0x00000044 msp: 0x00000048

Checking ROM masking
[DP32G0xx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc0000000 pc: 0x00000044 msp: 0x00000048
Erasing sector 0x07 = offset 0x0e00
Region cleared OK
Programmed up to 0x1000 (FLASH_ADDR=0xc400)
Checking ROM masking

ROM masking is set to 0b100. Setting ON...

Bootloader code programmed.
You can use uv_flash_fw to program main firmware, or just use stock tool to do it.

Checking ROM masking
[DP32G0xx.cpu] halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000000d4 msp: 0x200015d8
Erasing sector 0x77 = offset 0xee00
Region cleared OK
target/dp32g030.cfg:115: Error: could not read "k5_v2.01.27_fw": No such file or directory
in procedure 'uv_flash_fw'
in procedure 'write_image' called at file "target/dp32g030.cfg", line 315
at file "target/dp32g030.cfg", line 115

losehu, 2024-01-05:
> Here you have a ready to use config file for OpenOCD using STLinkv2 probe: https://github.com/Lar-Sen/Quansheng_UV-K5_Kitchen/blob/main/unbrick-toolkit/OpenOCD/target/dp32g030.cfg
I modified the cmd to：
@openocd -f interface/stlink.cfg -f target/dp32g030.cfg -c "reset halt" -c "uv_flash_bl bootloader.bin" -c "uv_flash_fw k5_v2.01.27_fw.bin" -c "shutdown"
After the burning was completed, my K5 turned into a brick

losehu, 2024-01-05:
> Here you have a ready to use config file for OpenOCD using STLinkv2 probe: https://github.com/Lar-Sen/Quansheng_UV-K5_Kitchen/blob/main/unbrick-toolkit/OpenOCD/target/dp32g030.cfg
> I modified the cmd [...]

Fortunately, I returned to normal after using my own cfg

ErikS-web, 2024-01-05:
Glad to read "you are _normal again_", what about your radio ?  ;-)

losehu, 2024-01-05:
> Glad to read "you are _normal again_", what about your radio ? ;-)

I am wondering if it is possible to debug code by configuring IDE and OPENOCD?

Lar-Sen, 2024-01-06:
Sorry for the inconvenience. My "1-hand" unbricking script wasn't tested with my recent updates to the target cfg file.
I preferred using telnet to interact with openOCD in daemon mode (-d option, port 4444)
Anyway I've been using this config many times on different handsets: for sure it works :)
Glad you were able ro recover yours. And got normal again ;)
For debugging you should take a look at GDB which works in tandem with openOCD. OpenOCD itself is not a debugging tool.

Lar-Sen, 2024-01-06:
My apologies again : The firmware file I provided in my repo was wrong! It was a full flash image, whereas 'uv_flash_fw' needs 0x1000+ region. Corrected.
This time it will work for good.